nion-software/nionswift #21: Add a zoom tool, zoom-in/out menu items when right clicking, regular menu items

cmeyer, 2017-08-17:
- Select zoom tool with hot key
- Zoom should occur around mouse click or mouse location when appropriate

Brow71189, 2020-05-26:
Maybe both ideas should be implemented? But I definitely think Tracy's idea should, because this is what the mouse wheel does in most applications. Right now it moves images up/down in Swift, which is not very useful.
A nice side effect of zooming around the cursor is that you can use that to move the image around. Moving images is often a bit awkward in many programs, so using "zoom around the cursor" is a decent workaround.

Niklas' idea might be useful, too, but I 'm afraid that this can be confusing for users. Maybe we should just get trackballs with mouse wheels?

ndellby, 2020-06-09:
I've looked, none of the good index finger trackballs come with scroll 
wheels, they only have the little minibuttons.

Niklas

Brow71189, 2020-06-09:
When I looked I found a couple. For example
https://www.amazon.com/dp/B01936N73I/ref=cm_sw_r_cp_apa_i_pT83EbD923DSJ

Or

https://www.amazon.com/dp/B002OOWB3O/ref=cm_sw_r_cp_apa_i_UU83EbK0H5M6V

Or

https://www.amazon.com/dp/B0050SPZMK/ref=cm_sw_r_cp_apa_i_HV83EbKDZC718

Or

https://www.amazon.com/dp/B07J3XZYGG/ref=cm_sw_r_cp_apa_i_.W83EbJTYQY61

Or

https://www.amazon.com/dp/B0735584RM/ref=cm_sw_r_cp_apa_i_IY83Eb66VPXCW

Actually I think having a scroll wheel would be nice also for general use
of the user pc.

jamesrussell216, 2024-08-14:
PR for this currently is: https://github.com/nion-software/nionswift/pull/1073

jamesrussell216, 2024-09-16:
After discussion on this issue we've got a proposed revised spec that @Tiomat85 is going to code too...

- One toolbar button for the zoom tool.
- Left-click to zoom in, modifier-left-click to zoom out, zooming around the mouse cursor position.
- Keep the +/- functionality; that should zoom around the centre of the image
- Leave the scroll wheel zooming functionality for now.
- Leave the left-click and drag to zooming functionality for now.

closed 2024-10-03 by cmeyer

Ion-e, 2024-10-09:
Implemented correctly, Thanks!

commit c2a389a3 (2024-10-03): Fix #21. Add zoom tool.