nion-software/nionswift #25: Line plot min/max calculation should sort data and exclude 5% on either end

cmeyer, 2017-09-03:


Brow71189, 2020-04-24:
Possible duplicate of #24? What does "sort data" mean? Does it mean it should use 5% - 95% percentiles (`numpy.percentile`) for calculating the display range?

cmeyer, 2020-04-24:
To figure out percentiles, you need to sort the data. `numpy.percentile` sorts the data for you.

This is different from #24 - This bug refers to how the data should be scaled statically; #24 refers to when the scales should be updated dynamically. For instance, during acquisition the data limits (5% and 95%) might be changing slightly. #24 suggests having some algorithm to only change when those limits go beyond a certain threshold of change.

Brow71189, 2020-04-25:
Understood. Although I'm not convinced that exluding 5% of data on the ends is a good idea. Usually line plots leave some extra white space.
Anyway, I'm gonna work on #24 first, since that is a big annoyance.

cmeyer, 2020-04-25:
Marking this as difficult due to performance considerations.