nion-software/nionswift #32: Snapshots should be automatically numbered

cmeyer, 2017-09-04:


cmeyer, 2018-10-09:
Allow numbering, sample info, etc.

Extended request by QR.

cmeyer, 2019-01-28:
From ND:

We could add an "automatic naming feature"
Any snapshot would be automatically named as:
Sample+location+sequence number+data type

Where location is named in the stage control, we already have this. Sample is the sample name. And sequence number just counts at that location.

So you would get names like:
SiN3-thin area-43-HAADF rather than "snapshot of haadf"

ndellby, 2021-03-24:
Remember Quentin's comment, decorations should be appended not prepended, Thus SiN-HADF should become "SiN-HADF-43-snap-FFT"
 rather than "FFT of snapshot of HADF SiN 43"
This sorts better and when text is truncated you don't get "FFT of Snapshot of..."

TomaSusi, 2021-03-25:
YES.

cmeyer, 2022-05-21:
Personal notes from 2021-09-03:

- Only used substitutions for dynamic content; not structure?
- {src-title}{processing}{r-var}
- "Ronchigram (FFT)"
- "src-title" is the source title or this items title if specified
- "processing" is the processing description slug
- "r-var" is the assigned ref var - treated separately.
- Q: How do I deal with the fact that processing occurs on a data item, but the title is associated with the display item?
- Q: For `src-title`, what title is used? The rendered title or just the `src-title` of the source?
- "Ronchigram (FFT) (Inverse FFT)"
- Overarching goal is to make it easier for the user to label the data.
- {device}{signal-type}
- "EELS Camera Spectrum"
- {signal}
- "EELS Spectrum"
- {collection-style}
- {dimensions}
- {acquisition-id}

cmeyer, 2023-03-22:
See also:

- #383

commit b22deae1 (2024-04-28): Fix #32. Add dynamic title support.
closed 2024-04-28 by cmeyer