nion-software/nionswift #37: Solve problem of accidentally creating new intervals when trying to edit existing

cmeyer, 2017-09-04:
From Tracy:
When making interval annotations on line plots, I never find myself wanting more than one, but often end up accidentally drawing several. Let's just make it default to one interval annotation unless you hold the special key that let's you draw a second one. This would prevent the pollution problem without removing potentially useful functionality.

See also #27

Brow71189, 2018-03-15:
That is also a common problem we have with line plots. One very helpful thing would also be that the cursor changes to this little two-ended arrow when you are going to change an existing interval. That would already help a lot in knowing where to position the mouse to do what you want.

Brow71189, 2020-04-25:
This is still one of the most annoying things when working with interval regions. I'd like to propose the following improvements:

1. A simple thing would be to increase the "tolerance" for clicking the edge of an interval graphic. Currently this is set to 4 px but I think we can definitely go for something like 8 here. Additionally I would like to change the cursor style to the two-sided arrow when it is going to change the width of the interval.

2. A general improvement for selecting graphics, especially when there exist multiple ones on a data item and they overlap or are close to each other: The `Graphic.test` function should also return some sort of a distance measure. Depending on the `part` it returns, it would calculate the appropriate distance of how far away a mouse click happened. So for example for an interval graphic, I would calculate the distance from the two vertical lines if the click happened next to them, and the distance from the horizontal line if the click happened somewher within the interval.
The containing `CanvasItem` could then sort the graphics according to this distance and select the one with the smallest distance to the mouse.
This should significantly improve handling multiple overlapping graphics.

3. Again specific to interval regions: Very often you create one if you just click on the image which has width 0 or something very small. I would suggest that we introduce a minimum width under which intervals don't get created by simply dragging over a line plot.

cmeyer, 2020-04-25:
These sound like good ideas - but I'd like to see how other software handles this too.

Brow71189, 2020-04-27:
Regarding selecting overlapping graphics (or objects in general), I checked out two pieces of software I'm using: Inkscape and Onshape.
In both of them it works pretty much as you would expect: You select the object that your mouse cursor is closest to. Inkscape makes this fairly simple because it only selects onjects when you hit their outlines. In Onshape you can select faces, edges, corners, etc. And there you simply just get whatever you're closest to. This feels pretty intuitive for me when I'm working with it.

cmeyer, 2020-08-15:
@Brow71189  Can we close this now or is there a reason for it to be still open?

Brow71189, 2020-08-15:
I think we can close this one.

closed 2020-08-15 by cmeyer