nion-software/nionswift #50: Add easier way to move zoomed-in images with the mouse

cmeyer, 2017-09-04:
Currently requires selecting the hand tool. Can this be a hot-key 'h' to switch to hand tool? What's the shortcut in Adobe products?

Brow71189, 2020-05-26:
One thing that works in different software products is that holding down the third mouse button (which is usually pressing the mouse weel) lets you move items around (independent of whether you are zoomed in or not).

ndellby, 2020-10-30:
we have 101 keys, and have lots of things we want to combine with mouse motion, is there no way that we can make this neater and integrated, we already have (creaky) S click, T click and we should have A click and B click. The sample map has contrl-click to move the stage. At present we have to change focus for key detection to work- can we do it better?

cmeyer, 2024-02-24:
Both 'h' and drag clicking work. Closing.

closed 2024-02-24 by cmeyer