nion-software/nionswift #55: Allow adjustment of rect by dragging on sides

cmeyer, 2017-09-05:
Either provide handles or allow dragging directly on the sides of the rectangle to reshape it by adjusting one side only (or holding down Alt and adjusting from the center).

jamesrussell216, 2025-03-27:
Just for reference what we're looking for are handles in similar positions to these...

[image]

That only change the dimension in the corresponding axis.

jamesrussell216, 2025-06-18:
We discussed this, the key question that came out of this is it is _actually_ needed? Is there a use case we're currently not sufficiently satisfying with dragging from the corners that warrants the functionality, or is this one just "development for development sake".

cmeyer, 2025-06-18:
It is very difficult (impossible?) to keep one dimension fixed when adjusting corners. I don't remember the specific circumstances, but this has come up multiple times for me over the years.