nion-software/nionswift #82: Export options for display vs data, sequences, and graphics burn in

cmeyer, 2017-09-23:
See #81.

tcpekin, 2019-05-16:
I would highly support the ability to export images with a scale bar burnt in. This makes making quick graphics very easy.

Brow71189, 2020-09-04:
@cmeyer Can you assign me please? This looks like something I could work on when I have some time for general Swift development.

cmeyer, 2020-09-04:
I added the planning label. Unfortunately, this is one of those issues that I quickly recorded without thinking through the full plan. It might be better to create new issues with specific details.

For example, when "burning in graphics", should the user specify a DPI? I would encourage users to use SVG export instead and then use a separate graphics program to render the text, which gives them significantly better control and results. Also, the font rendering in Swift is limited in quality and improving it might be a precursor to this feature.

Or are you thinking of something regarding the display vs data export? That's more clear to define and maybe we can open another issue to be clear about what we're doing?

Ion-e, 2024-06-18:
https://github.com/nion-software/nion-internal/issues/112 this ticket has been closed as it's a duplicate of this one.