nion-software/nionswift #111: Provide easier way to discover keyboard shortcuts

cmeyer, 2018-02-28:
From Niklas December 21 2017:

2 keyboard shortcuts are great, if you can find them. I knew that the computation window has moved to a keyboard shortcut, it took me 15 attempts to come up with cntl-r for computation. Right clicking on a window should give access to a list of options.

cmeyer, 2024-08-16:
See this link for organization categories; consider making dialog like this in anticipation of eventual shortcut editor:

Organizing into groups https://helpx.adobe.com/illustrator/using/default-keyboard-shortcuts.html

Possible dialog ideas in anticipation of eventual editing capabilities (takeaway: organize into groups with combo box selecting which group is displayed):

[image]

cmeyer, 2024-08-16:
From Niklas December 21 2017:

> 2 keyboard shortcuts are great, if you can find them. I knew that the computation window has moved to a keyboard shortcut, it took me 15 attempts to come up with cntl-r for computation. Right clicking on a window should give access to a list of options.

lisham2000, 2024-08-20:
Hi Chris, just following up on the pull requests discussion for the key bind shortcuts work.  The purpose of this work was to provide a relatively quick solution to the problem proposed at M&M which was easily accessible common keyboard shortcuts for beginner users. With this in mind and with discussion with the team/James, we thought it best to make the proposed solution a non-modal dialog such that it could run concurrently with swift windows without dialog persistence causing an issue as well as having flexibility to print it if new users wanted a copy to refer to. Similar examples would be seen in most software like this one from adobe illustrator https://helpx.adobe.com/content/dam/help/en/illustrator/pdf/Ai_cheat-sheet_Windows.jpg or this one from GitHub that  we use https://education.github.com/git-cheat-sheet-education.pdf This would be separate to a shortcut editor and manager similar to the one that you proposed from illustrator which I think is a much larger work piece of work compared to the couple of hours or so that this took me as I think you’d want to in some way tie it to the project or a profile system as multiple people will use swift from the same machine so you would want to keep individuals key binds specific to them to save people having to redo shortcuts as soon as they log onto a shared. I’m aware that the code itself would need to be changed to maintain encapsulation and contain the inter code shortcuts and maybe it would be more useful to name it “default keyboard shortcut identifier” or something alike as to separate the problems defined in #111 vs #298.

lisham2000, 2025-01-03:
[new_key_config.json](https://github.com/user-attachments/files/18299180/new_key_config.json)

cmeyer, 2025-01-07:
Here is the text of the attached file

```
{
    "General Window Operations": [
        {"key": "Ctrl + S", "description": "Save display snapshot."},
        {"key": "Ctrl + D", "description": "Duplicate item."},
        {"key": "Ctrl + \\", "description": "Toggle filter."},
        {"key": "Ctrl + Shift + K", "description": "Assign variable reference."},
        {"key": "Ctrl + Shift + U", "description": "Copy UUID."},
        {"key": "Ctrl + Shift + R", "description": "Open data item recorder."},
        {"key": "Ctrl + E", "description": "Edit computation."},
        {"key": "Ctrl + T", "description": "Edit title."},
        {"key": "Ctrl + Shift + E", "description": "Edit data item script."},
        {"key": "Ctrl + R", "description": "Run scripts."},
        {"key": "Ctrl + K", "description": "Open console."},
        {"key": "Ctrl + [ / Ctrl + ]", "description": "Navigate workspaces (previous/next)."},
        {"key": "Ctrl + Alt + L", "description": "Create new workspace."},
        {"key": "Ctrl + Alt + E", "description": "Add ellipse graphic."}
    ],
    "Display Panel Operations": [
        {"key": "Delete", "description": "Delete graphics."},
        {"key": "V", "description": "Cycle through display modes."},
        {"key": "Alt + Shift + F", "description": "Toggle frame rate display."},
        {"key": "Alt + Shift + L", "description": "Toggle latency display."}
    ],
    "Tool Modes": [
        {"key": "E", "description": "Set tool mode to pointer."},
        {"key": "H", "description": "Set tool mode to hand."},
        {"key": "Z", "description": "Set tool mode to zoom."},
        {"key": "N", "description": "Set tool mode to line."},
        {"key": "C", "description": "Set tool mode to rectangle."}
    ],
    "Raster Display Operations": [
        {"key": "0", "description": "Fit view to content."},
        {"key": "Shift + 0 / )", "description": "Fill view."},
        {"key": "1", "description": "View preset 1."},
        {"key": "2", "description": "View preset 2."},
        {"key": "Enter", "description": "Auto-display raster."},
        {"key": "Arrow Keys", "description": "Nudge graphics in corresponding direction."},
        {"key": "Shift + Arrow Keys", "description": "Move graphics in corresponding direction."},
        {"key": "Ctrl + Arrow Keys", "description": "Adjust slice nudge."},
        {"key": "-", "description": "Zoom out."},
        {"key": "= / Shift + = / +", "description": "Zoom in."}
    ],
    "Line Plot Operations": [
        {"key": "Control + Hover (Horizontal Axis)", "description": "Cursor becomes split-horizontal."},
        {"key": "Control + Hover (Vertical Axis)", "description": "Cursor becomes split-vertical."},
        {"key": "Control + Hover (Interval/Channel Start/End)", "description": "Cursor becomes size-horizontal."}
    ],
    "Graphic Adjustments": [
        {"key": "Shift + Line Drawing", "description": "Snap line to specific angles."},
        {"key": "Alt Modifier (Interval Graphic)", "description": "Resize from center of interval."},
        {"key": "Control Modifier (Interval Graphic)", "description": "Move entire interval."},
        {"key": "Shift + Point Adjustment", "description": "Constrain point movement to one axis."},
        {"key": "Shift + Rectangle Rotation", "description": "Constrain rotation to 45° increments."},
        {"key": "Shift + Rectangle/Square Adjustment", "description": "Constrain shape to square."}
    ],
    "Selection Operations": [
        {"key": "Shift + Click", "description": "Extend selection."},
        {"key": "Control + Click", "description": "Toggle item selection."},
        {"key": "Control + Click (Display Panel)", "description": "Toggle display panel in selection."},
        {"key": "Shift + Click (Display Panel)", "description": "Add display panel to selection."},
        {"key": "Control + Click (ImageCanvasItem)", "description": "Add item to graphic selection."}
    ],
    "Tab Navigation": [
        {"key": "Control + 1", "description": "Switch to first tab."},
        {"key": "Control + 2", "description": "Switch to second tab."}
    ],
    "Console Operations": [
        {"key": "Control + C (MacOS)", "description": "Add new prompt text."}
    ]
}
```

jamesrussell216, 2025-05-19:
Outputs from @lisham2000 work in this PR...

https://github.com/nion-software/nionswift/pull/1465

cmeyer, 2025-07-26:
Offline discussion:

- the keyboard shortcuts dialog is going to be difficult to pull off since it contains many contexts that will be difficult to summarize in a single window
- keyboard shortcuts appearing in the menus do not need any more detail as they are already discoverable
- keyboard shortcuts associated with existing UI elements should have tooltips, e.g. rectangle tool
- keyboard shortcuts associated with display panel should be discoverable by mouse-over or something similar a corner of the display panel
- keyboard shortcuts associated with a mode such as dragging should appear while dragging, similar to a tooltip
- a central manager should centralize shortcuts where possible (#1465)

cmeyer, 2025-07-26:
Added a sub-issue to break up the work of this issue.