nion-software/nionswift #113: Provide an easy way to quickly copy and paste a crop to a new data item

cmeyer, 2018-02-28:
Provide for 1D, 2D, 3D, and 4D data.

cmeyer, 2018-10-09:
Request at workshop 2018.

Brow71189, 2020-12-17:
I think this would be useful as a general tool, not only for the "Crop" computation. I think wanting to apply a computation that you already set up for a certain data item to another, similar data item is something very common.

One complication might be that we need to replicate all the input UI elements to set up the copied computation. I guess this is already possible but a bit complicated to achieve. To simplify we could seperate out the setup code that is usually happening in the MenuItem class into its own function/object which takes the start values for all the parameters as input arguments. Like this, the "copy" function could call this object and pass in the parameters from the computation that is about to be copied as start values. Like this we use the same code to set up the copied computation as we use to set up the initial computation.

Another option would be to have the computation class supply a `copy_to_new_input_item` method. This would limit the copy capability to computations that explicitly support it, but it would make the copy operation itself trivial. It could also help solving some problems with replicating the input UI elements of the source computation, especially when they are spread over multiple input data items.
As an example, consider a computation that uses a mask that is set up as a separate data item and processes another data item utilizing this mask. If a user now selects only one new data item and attempts to copy the aforementioned computation, we need to decide how to handle the missing second input data item. From a user's perspective the intended behavior would be that the copied computation uses the same mask than the original one but applies it to the selected data item. An external "copy" function has no way to know how to handle this case because it lacks knowledge about what the computation is doing. An explicit "copy" implementation in the "Computation" object would have this knowledge and could make sure this case is handled properly.

cmeyer, 2021-12-21:
See also:
- #178