nion-software/nionswift #149: Allow graphics behind other graphics to be selected

cmeyer, 2018-09-10:
Either by cycling through layers on each click (Andrew) or by keyboard shortcut to select item under or over (Matt). Investigate how other applications (CAD, Illustrator) handle it.

Ion-e, 2024-09-05:
In Illustrator, there’s a dedicated UI for managing layers, where you can track and mofidy rules for each layer such as opacity, locks, masks, order...(see: https://helpx.adobe.com/illustrator/using/layers.html). However, layers in Illustrator often contain multiple objects, making it more complex that we would need for the requirements here. My understanding is that Swift already has a certain degree of the layers feature, albeit not directly controlled by the user (a bottom layer for data and a top layer for graphics).

My understanding in Illustrator there are 2 methods (besides the layers) to select overlapping objects in the same layer (See: https://helpx.adobe.com/illustrator/using/selecting-objects.html):

Method 1: Control+Click to iterate between them (see: "Select Behind Objects" in previous link).
Method 2: Right click the object > choose "Select" > "Next Object above/Below" (See: "Select the next object" in previous link)

If we decide to go for a "layer per graphic" solution, the work on #58 may have some unintended behaviours with this feature.

Ion-e, 2024-09-12:
@cmeyer

jamesrussell216, 2025-05-21:
I think this one needs a half hour design chat at some point. I'd like to see if we can do something simple to solve the problem without coding up a chunk of Illustrator ;-)

Ion-e, 2025-07-18:
For this ticket. Do we only allow to iterate through the graphics by using Tab key? Was another functionality introduced? @cmeyer