nion-software/nionswift #153: Exporting live data should snapshot it first so uuid doesn't get confused when reimporting.

cmeyer, 2018-10-06:


Brow71189, 2018-12-21:
Maybe one could also do it the other way round: Assign a new uuid to important items. The reason I'm proposing this is that it would also help avoiding uuid confusion when importing a data item twice.

cmeyer, 2018-12-21:
Do you mean while importing? Snapshot already assigns new uuid.

Brow71189, 2018-12-21:
Yes I mean you would assign a new uuid during import. I'm not sure if this creates other issues. But the uuid confusion when you import the same data multiple times is also really annoying.
I had this specific issue when working with files exported in raw numpy format. I wanted to keep the calibrations, so I just edited the .npy file and imported it again (and did that multiple times). Of course all of these data items ended up having the same uuid.

cmeyer, 2025-03-20:
Note: see how this is handled in nhdf.