nion-software/nionswift #199: Add support for copy/paste to add a layer to line plot.

cmeyer, 2019-01-05:


meyer9, 2019-08-06:
First of all, there seems to be other problems that need addressing before this. For example, you can't currently copy/paste a display item anywhere.

Second of all, I don't think copying and pasting to add a layer is the expected behavior. If you look at something like Photoshop or Illustrator, they have a richer layer dialog that allows you to edit the order of the layers, duplicate layers, etc. If anything, I think that's probably the correct approach here.

[image]

cmeyer, 2019-08-06:
Let's do it like this for now: add a menu item called "Copy Display" which in the context of this issue will create MIME data with the `MimeTypes.DISPLAY_ITEM_MIME_TYPE` data (see code for dragging and dropping a display item). Then if 'paste' occurs with a display item selected that is a line plot, and the source display item is compatible with the target display item (see drag and drop again), then paste it. So the user can "copy display" menu item and the "paste" menu item to a new line plot.

cmeyer, 2019-08-19:
18fc15304db1ea4ee9aaa34dd137e7d665818bcb

closed 2019-08-19 by cmeyer