nion-software/nionswift #206: Make created, modified, and data timestamp more accessible in UI.

cmeyer, 2019-01-12:


jamesrussell216, 2024-07-24:
On this one @cmeyer - in the discussion is the prevailing opinion that we should just use the original data object acquisition date _everywhere_ and accept for "live" data objects that the easiest way to find those is going to be using the collections panel?

jamesrussell216, 2025-01-06:
@cmeyer did we get to any solid conclusions on this one can you remember? Simplicity is a currency all of it's own, and I'm inclined to agree with the above. Particularly with the improvements you've made to the collections panel; live data items are very easy to find in there.

cmeyer, 2025-06-06:
From @Ion-e 

> The image Date section in the inspector doesn't update with live items. As title, also please note that the "Date" info in the inspector it's affected by the same behaviour although unsure about how would we want the latter to behave.

cmeyer, 2025-06-06:
From @jamesrussell216 

> Multiple Data Items getting wrong timestamp.
> When you make a new "Multiple Data Items" display item, the time stamp you get in the Data panel is the time stamp from one of the data items, not from when the Multiple Data Item was made. Slightly oddly it does show up correctly in the Project Items window though...