nion-software/nionswift #217: Combining line plots should keep the original names as the layer names

cmeyer, 2019-02-06:
ND: If I have two spectra I want to compare, and I name each spectrum and then drag one into the other Swift loses any information about the spectra, the data become data0 and data1.
So the Layer 0 and Layer 1 name boxes should take the name of the source spectrum.

dhauge, 2021-07-05:
@cmeyer Just to clarify, for a line plot, looks like there are two notions of line plot name, one the line profile name shown in the title of the window, and the second the name of the layer in the `Line Plot Display Layers` section. The latter is not set when there is only one line plot in a display. However, if I do specify the layer name in each of the individual line plots, and drag one into another, the layer name of the latter is retained but the first is lost?

Is the intent of this issue to use the line profile title as the name of each display layer? Or is it just to make sure that if the layers of the individual plots are specified, that name is not lost when one line plot is dragged into another?

ndellby, 2021-07-07:
The only UI that's easily accessible for a line plot is the title, so the obvious desired functionality is : if the data series has been named then when two plots are merged the name of the layer for each plot becomes the name of the layer in the combined plot. But if the layer is unnamed then the title is used to name the layer in the combined plot. That way one can easily name the individual data merge them, get the names in the right places and then rename the combined plot as desired.

cmeyer, 2021-07-08:
My thinking on the titles was that a data item would always have a title. However, since a single display item may display multiple data items, the display item itself may have a title. In addition, the role that a data item plays within a display item (layer) may be different from the data item's original purpose. For instance, a data item might have a specific name when displayed by itself; but a general name, e.g. "reference spectrum", when displayed within a composite display item. So I see three titles involved: the data item title, the display item title, and the display item layer label.

1. The display item would use the data item title by default; but if the display item title was set, it would display that instead. This is dynamic in that if you change the underlying data item title, the display item title may change.
2. The display layer would use its label in all cases. The label is not dynamic. Once it is set, it is not changed except by the user.
3. The display layer label would initially be set to whatever was displayed for the display item title in step 1 with a pound sign number attached to it if the incoming data item had no title or else the title of the data item.

So when you dragged in a new data item, it would set the layer label to the title of the data item if not empty; or the used title of the display item with pound sign layer number appended if the data item had no title. This is close to https://github.com/nion-software/nionswift/issues/217#issuecomment-876004582 with the added clarification that display layer labels are not dynamic.

There are a number of open issues to have dynamic titling of dependent data items and display items. So these heuristics may change in the future.

dhauge, 2021-07-08:
@cmeyer Are you sure you don't want the display layer labels to continue to be dynamic if not explicitly set? I'd think it would be nice to have the displayer layer name track the data item name if the former is not explicitly set, using logic similar to what is there now but using the image name rather than `Data` in the dynamically generated name.

cmeyer, 2021-07-09:
Sure - that would work too. The only caveat is that the titles are not set up to automatically update. That's a separate larger project. Let's chat about this on Monday.

jamesrussell216, 2025-03-11:
https://github.com/nion-software/nionswift/pull/1359

https://github.com/nion-software/nionswift/pull/1360

closed 2025-03-11 by cmeyer

Ion-e, 2025-03-27:
Verified as implemented, thanks!