nion-software/nionswift #218: Can't access data item form a channel with a reference key

kociak, 2019-02-08:
Hi

I was trying to get the channel and data item from a camera using the example script given at: https://nionswift-instrumentation.readthedocs.io/en/latest/cameras.html#how-do-i-find-data-items-associated-with-viewing-and-sequence-acquisition

that I have adapted to a script to be run:

```
from nion.typeshed import Interactive_1_0 as Interactive
from nion.typeshed import API_1_0 as API
from nion.typeshed import UI_1_0 as UI
import numpy as np
from scipy import optimize
from nion.utils import Registry

def script_main(api_broker):
    interactive = api_broker.get_interactive(Interactive.version)  # type: Interactive
    api = api_broker.get_api(API.version, UI.version)  # type: API
    stem_controller = Registry.get_component("stem_controller")

    ronchigram = stem_controller.ronchigram_camera

    reference_key = ronchigram.make_reference_key()

    print(help(api.library.get_data_item_for_hardware_source))
    print(ronchigram.hardware_source_id)

    data_item = api.library.get_data_item_for_hardware_source(stem_controller.ronchigram_camera)
```
I get the following error: `AttributeError: 'CameraHardwareSource' object has no attribute '_hardware_source'`

Any help welcome!

cmeyer, 2019-02-08:
When using Swift, there are two types of objects: "native" and "API". Generally the two can't be mixed. In this case, the `ronchigram` is a native object and the `api.library.get_data_item_for_hardware_source` is expecting an API object. This is confusing and I'm planning on simplifying this in future versions.

For now, you can convert from a native object to an API object using `api.get_hardware_source_by_id`. See the example below:

```
from nion.utils import Registry

def script_main(api_broker):
    api = api_broker.get_api(version="1.0", ui_version="1.0")
    stem_controller = Registry.get_component("stem_controller")
    ronchigram = stem_controller.ronchigram_camera
    ronchigram_hardware_source = api.get_hardware_source_by_id(ronchigram.hardware_source_id, "~1.0")
    data_item = api.library.get_data_item_for_hardware_source(ronchigram_hardware_source)
    print(data_item.title)
```

cmeyer, 2019-02-08:
Closing this as resolved (see commment above). Also filed additional bug to provide improved documentation regarding this issue. #219.

There may be other similar situations you encounter. If so, please file additional bug reports.

closed 2019-02-08 by cmeyer