nion-software/nionswift #242: Calculating the inverse FFT of an FFT of real data should produce real data again.

Brow71189, 2019-03-28:
To see the problem, take any image, run "FFT" computation on it and then "inverse FFT" on the result. The end result will still be complex and will be displayed as the logarithm of the absolute of the result. What you should get is the same image as the one you started with.

4kbt, 2020-08-09:
I've had a gander at the code today, planning to unit-test and bisect to understand/rectify the Issue, but found the computational workflow difficult to follow. Is there a summary design-doc for the computational framework I could consult?

cmeyer, 2020-08-09:
Unfortunately, I don't have a good design document for the computational workflow. Please file an issue to request that to be written. A rough description is that all processing is confined to the `niondata` framework. So if it is fixed there, then Swift itself will utilize the fix. You can import `niondata` directly and manipulate data via the `xdata_1_0` interface and there is no need for Swift to be involved.

Solving this issue (FFT - Inverse FFT) isn't going to be easy to fix this problem as it will require tracking the type of complex data (i.e. whether it is Hermitian or not) that results from a particular operation (forward FFT). There is no mechanism to record that property. We could consider temporarily using a top-level metadata value, but my approach so far has been that metadata is not descriptive of the data and does not participate in processing functions. Even after tracking the Hermitian property, it's also problematic to figure out which operations break that property (i.e. force the complex data to be fully complex).

Are you running into this issue in the real world? If so, a workaround would be to explicitly use a different inverse FFT function when you were sure the data was Hermitian.

4kbt, 2020-08-10:
I'm not hitting this in the wild.  I'm a grad-school friend of Tracy's who was curious how far Nion's open-source codebase had come over the years.  When I saw that Swift was GPL-v3 and had an organized list of Issues, it was very tempting to want to learn more. 

One of the best ways to learn more about a codebase is to find a small bug and make a contribution. I do a lot of fourier work in my gravitational experiments, so this Issue caught my eye as potential low-hanging fruit.

While I'm still learning about the Nion datastructures, it seems that addressing the original Issue might not require adding a metadata property.   It appears to ask that IFFT( FFT( data ) ) == data . One might not need to track that FFT(data) is special in any way before ensuring that IFFT( FFT(data) ) recovers the expected output?

I'll clone niondata later this week to learn more. Thanks!

cmeyer, 2020-08-10:
I'm excited to have you interested in the FFT's - and it might indeed be low hanging fruit.

I think the round trip is consistent, other than the result data type is complex and some floating point rounding errors resulting in non-zero imaginary components ~ 1E-14 for 32-bit floats. This has an explicit test: https://github.com/nion-software/niondata/blob/master/nion/data/test/Core_test.py#L73

We could do a check on the inverse FFT to see if the imaginary part is nearly zero (all values < 1E-13?). But that would affect performance so I'm hesitant to make that the default.

There is also a processing routine 'Convert to Scalar' which might do the trick. Unfortunately, it converts the display version of the data (i.e. scaled) not the underlying raw data. I'm partially tracking that problem is #39.

You might also be interested that NumPy has added improved scaling choices for FFT/IFFT. I haven't had a chance to evaluate if it works correctly or consistently. In `niondata` the forward scaling is `scaling = 1.0 / numpy.sqrt(data_shape[1] * data_shape[0])`. This was chosen to keep the RMS constant between real space and Fourier space data. See https://github.com/nion-software/niondata/blob/master/nion/data/test/Core_test.py#L91

I'd be interested to know what scaling is used when doing gravitational experiments.

On another side note: I think Swift would generally be useful in the astronomy domain also. I'm not sure if that's what you had in mind, but it might be an interesting side project for me (outside of Nion) to help structure an astronomy-specific package for Swift. I think in the long run, a few domains outside of STEM microscopy would help improve overall capabilities.