nion-software/nionswift #245: CSV export of 3d and 4d data should not be allowed

Brow71189, 2019-04-15:
CSVImportExportHandler should not try to export 3D or 4D data. Numpy savetxt cannot handle it so you get the following exception:
```numpy.savetxt(path_str, data, delimiter=', ')

"Expected 1D or 2D array, got %dD array instead" % X.ndim)

ValueError
:
Expected 1D or 2D array, got 4D array instead```

jamesrussell216, 2024-05-21:
Hi @Brow71189 we're looking at a bunch of the export issues at the moment. Just looking at this one - is the intention with this issue just to block off the ability to export 3d or 4d data, rather than making coding changes to support that data export? Would being able to do a csv export of 3/4d data be useful for a user?

Brow71189, 2024-05-21:
Yes, the intention here is to block off the ability to export 3D or 4D data to csv. The reasons for this are that 

1. It is not clear how to represent 3d/4D data in tabular form in a text file (would require some non-standard formatting that only custom readers can interpret)
2. Typically these files are too big to sensibly fit into a text file. With text files, you often want to load them with a text editor and have a quick look. With huge files, many text editors choke and hang which causes a poor user experience.

At least these were our initial thoughts. If you see a use-case for 3d/4d csv export, we should probably move that to a new issue and just disallow this type of data for the current exporter.

jamesrussell216, 2024-05-21:
Simple is a currency all of it's own! We'll get that sorted.

I would think you'd just have to split a 3D (and 4D) export into a 2D export across multiple separate files. But what the use case for this is I don't know; you're probably into the realms trying to do analysis in a separate software package but in that case I wouldn't have thought you'd want csvs anyway!

commit 5e0839db (2024-06-19): Fix #245. Added a check for no more than 2 dimensions in the data for CSV exports
closed 2024-06-19 by cmeyer