nion-software/nionswift #246: Allow creating data_item connections from API

Brow71189, 2019-04-16:
It would be great to have a way to specify connections between data_items from plugins and scripts, so that you get the little thumbnails that you get from computations.

cmeyer, 2025-03-20:
@Brow71189 Can you give your specific use case for this?

Brow71189, 2025-03-21:
This is 6 years old, so I cannot remember the exact application I had in mind back then. But I still think it would be nice if you could show relations between data items that  you create by a script or a plug-in.

cmeyer, 2025-03-21:
Yes, I agree - I just want to make a specific use case. The ones I know about are: computations; composite line plots and future composite raster images; and items related by acquisition, which will be covered by acquisition objects. Are there others you can think of?

Alternatively, what kinds of API would this be? 'add_relationship_between_items(relationship_type, items)' ?