nion-software/nionswift #254: spectrum image hangs then cannot be opened

AEljarrat, 2019-06-11:
Dear all, 

I thought this issue was worrisome enough it should be reported, although it is probably hard to replicate and it seemingly fixed itself in the end. It appeared using the pick average tool for a spectrum image. The tool hanged and the console was flooded with a repeating error message (see attached file). I stopped the process and reopened Nion Swift. When reopened, all my spectrum images (not only the one I was looking at the beginning) where called "Untitled" and could not be opened. I also noticed an error in the console (see attached file, starting from line 6654). By chance I found the issue was solved by closing the console and reopening the data. 

[error.txt](https://github.com/nion-software/nionswift/files/3275214/error.txt)

Best regards, 
Alberto.

cmeyer, 2019-07-22:
From this line in the log: `OSError: [Errno 98] Address already in use`, it looks like Swift had been launched twice. This can sometimes happen inadvertently if Swift doesn't shut down properly as a result of a camera not closing properly. And HDF5 within Swift doesn't handle double launches nicely.

Were you running on a microscope system or just offline?

I added a new issue to track the double-launch problem. #264

AEljarrat, 2019-07-23:
Hi Chris, 
to answer your question: I was running offline. 
If you already open a new issue, I guess I can close this one. 
Thank for looking into this.
Best regards,
Alberto.

closed 2019-07-23 by AEljarrat