nion-software/nionswift #256: Nion crashes when used for too long

laenglem, 2019-06-26:
I use Debian 9. Nionswift crashes when I go thorugh too many images. Especially when using the Thubnail Browser. Please help me because it is really annoying.

Here is what is says on the Terminal.

user@device:~$ nionswift
Python version 3.7.3 (default, Apr  3 2019, 05:39:12) [GCC 8.3.0]
User interface class QtUserInterface / PyQtProxy
Application data: /home/manuel/.config/Nion/Nion Swift/nionswift_appdata.json
Loading plug-ins from /home/manuel/.local/share/Nion/Nion Swift/PlugIns
NOT Loading plug-ins from /home/manuel/Dokumente/Nion/Swift/PlugIns (missing)
Plug-in 'nionswift_plugin.DM_IO' loaded (nionswift_plugin).
Plug-in 'nionswift_plugin.TIFF_IO' loaded (nionswift_plugin).
Plug-in 'nionswift_plugin.none' loaded (nionswift_plugin).
NOT Loading color maps from /home/manuel/.config/Nion/Nion Swift/Color Maps (missing)
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Using existing document /home/manuel/Dokumente/VOID-Data/Platinum/20190521-Hel-33-Pt+-80eV-dose15/Nion Swift Library 13.nslib
Welcome to Nion Swift.
Using existing document /home/manuel/Dokumente/VOID-Data/Krypton/20190424-Hel-37-Kr+-100eV-dose15/Nion Swift Library 13.nslib
Welcome to Nion Swift.

(python3:8621): GLib-ERROR **: 14:50:55.313: Creating pipes for GWakeup: Zu viele offene Dateien
Trace/Breakpoint ausgelöst

cmeyer, 2019-06-26:
Thanks for reporting this issue. This looks like an issue with too many threads being created while generating thumbnails. I will investigate the problem to see if I can reproduce it and fix or at least give you something else to try.

cmeyer, 2019-06-26:
This problem is specific to PyQt UI. It is caused because the PyCanvas not being deleted properly, so the rendering threads are never released. I'm exploring fixes.

@laenglem Is it possible for you to installed `nionswift-tool` and use it to launch Swift? `nionswift-tool` is the C++ UI back-end for Swift. It does not have this problem.

Depending on your environment, you should be able to do:

`conda install -c nionswift-tool`

Then launch with

`nionswift`

laenglem, 2019-06-27:
Hi,
I used nionswift-tool and now it works perfectly!
Thanks!

cmeyer, 2019-06-27:
Problem is fixed in PyQt case: d6d89d532a56a860bcc8b0aeb9f3d7c91a3a2639. Will be released in 0.3.24.

closed 2019-06-27 by cmeyer