nion-software/nionswift #258: Allow Fourier masks to be float instead of binary

cmeyer, 2019-06-27:


MattRoyle, 2026-09-28:
Notes from backlog refinement 2026/09/22:
What - Floating point masks that allow transitions between the occluded and included, 0 and 1 regions of the mask.
Why – allow non-sharp edges on the masks to minimize artifacts from masking – discussed in a recent meeting as a desired feature. This would be a prerequisite to this feature.

From @KRLango: The masks are used to select specific frequencies in the fft to turn back into an image by multiplying the fft by the mask and then performing the inverse Fourier transform on that. If you have sharp edges in the fft, they create ripples around objects (called ringing). The common solution to that is to have a "transition" at the edge of the mask where you smoothly go to 0, this requires masks to have values between 0 and 1. Once float masks are implemented, providing a means of using mask transition functions can be further discussed.

Pitfalls - There are a bunch of functions that combine masks that currently all assume the mask is Boolean that will need to be updated to be floats (or at least tested to ensure they work with float masks)

cmeyer, 2026-09-28:
We can figure out the mathematics/niondata-functions behind this before connecting it to real graphic masks.