nion-software/nionswift #260: menu edit/preferences does fail with AsertionError

foetrix, 2019-07-01:
Hi,
When I try to open the edit/preferences menu nionswift fails with this error:

`Using Python from command line: /home/foetrix/miniconda3/envs/nionswift
Python version 3.7.3 (default, Mar 27 2019, 22:15:47) [GCC 7.3.0]
User interface class QtUserInterface / HostLibProxy
Application data: /home/foetrix/.config/nionswift_appdata.json
Loading plug-ins from /home/foetrix/.local/share/Nion/Nion Swift/PlugIns
NOT Loading plug-ins from /home/foetrix/Dokumente/Nion/Swift/PlugIns (missing)
Plug-in 'nionswift_plugin.DM_IO' loaded (nionswift_plugin).
Plug-in 'nionswift_plugin.TIFF_IO' loaded (nionswift_plugin).
Plug-in 'nionswift_plugin.none' loaded (nionswift_plugin).
NOT Loading color maps from /home/foetrix/.config/Color Maps (missing)
Using existing document /home/foetrix/Dokumente/Nion Swift Library/Nion Swift Library 13.nslib
Welcome to Nion Swift.
Traceback (most recent call last):
File "/home/foetrix/miniconda3/envs/nionswift/bin/NionSwiftLauncher/bootstrap.py", line 24, in bootstrap_dispatch
return getattr(object, method_name)(*args)
File "/home/foetrix/miniconda3/envs/nionswift/lib/python3.7/site-packages/nion/ui/QtUserInterface.py", line 1663, in triggered
self.trigger()
File "/home/foetrix/miniconda3/envs/nionswift/lib/python3.7/site-packages/nion/ui/QtUserInterface.py", line 1658, in trigger
self.on_triggered()
File "/home/foetrix/miniconda3/envs/nionswift/lib/python3.7/site-packages/nion/swift/DocumentController.py", line 996, in open_preferences
preferences_dialog = PreferencesDialog.PreferencesDialog(self.ui, self.app)
File "/home/foetrix/miniconda3/envs/nionswift/lib/python3.7/site-packages/nion/ui/PreferencesDialog.py", line 119, in __init__
rebuild()
File "/home/foetrix/miniconda3/envs/nionswift/lib/python3.7/site-packages/nion/ui/PreferencesDialog.py", line 114, in rebuild
change_selection({selected_index})
File "/home/foetrix/miniconda3/envs/nionswift/lib/python3.7/site-packages/nion/ui/PreferencesDialog.py", line 70, in change_selection
assert 0 <= index < len(preference_pane_delegates)
AssertionError
`
Cheers,
foetrix

cmeyer, 2019-07-01:
The problem is when there are no preference dialog entries.

foetrix, 2019-08-30:
Sorry for the late reply,
you are right, my bad.
even it would be nice not getting an error if no plugins are installed.

closed 2019-08-30 by foetrix