nion-software/nionswift #261: nionswift can't open old libraries

laenglem, 2019-07-03:
When trying to open a library created in Jannuary or May with the updated nionswift it does not show anything, the library is empty. The text from the terminal:

Migrating library: /home/manuel/Dokumente/VOID-Data/Manuel/20190121-Hel-13-Ar+-100eV-dose15/Nion Swift Library 12.nslib -> /home/.../libraryname/Nion Swift Library 13.nslib
Using existing document /home/.../libraryname/Nion Swift Library 13.nslib
Welcome to Nion Swift.

cmeyer, 2019-07-03:
Is it possible to upload the original library to Drop Box or Google Drive or something similar? You can email me the link directly (don't post here if you don't want it to be public).

cmeyer, 2019-07-09:
I've released a new version 0.14.6 -- can you update to this version and try it again?

Usually you can use something like this to update (from the conda environment where Swift is installed):

`conda update -c nion --all`

cmeyer, 2019-12-24:
Closing as fixed.

closed 2019-12-24 by cmeyer