nion-software/nionswift #267: Examples for interactive scripting cause error

MEisebitt, 2019-08-12:
The examples provided here:

https://nionswift.readthedocs.io/en/stable/api/interactive.html

causes a error, even if you include numpy additionally.

Was on Windows 7

cmeyer, 2019-08-12:
Partially fixed a4a7e78ef6076b22d50cc7ece850e4a702c41aba

I updated the example. The new version doesn't exactly roll off the tongue -- so I created #268, a request to add a new API for this.

It will take a few minutes before this documentation change makes it to the 'stable' documentation on readthedocs. Until then, here is the updated code example:

```
def script_main(api_broker):
    interactive = api_broker.get_interactive(version='~1.0')
    api = api_broker.get_api(version='~1.0')
    is_confirmed = interactive.confirm_yes_no('Are you ready?')
    if is_confirmed:
        print('Proceeding...')
        data_item = api.library.create_data_item_from_data(numpy.random.randn(16, 16))
        api.application.document_windows[0].display_data_item(data_item)
```

closed 2019-08-12 by cmeyer

MEisebitt, 2019-08-12:
Nice, thanks for the quick response.

MEisebitt, 2019-08-13:
Actually it still reads: 

File "Ex.py", line 1, in <module>
File "Ex.py", line 9, in script_main
AttributeError: 'API' object has no attribute 'show'

Is this planned, due to #268 ?