nion-software/nionswift #275: Multi line plot updates incorrectly when input data item x-calibration changes

cmeyer, 2019-09-15:
Emails from 2019-04-05

AM:
The problem exists only with stacked line plots. So I think it is a problem with how the display range for the x-axis is calculated for these display items: For the case we have here, where the calibration offset changes, the other line plot (the one that is not live) stays in a different position, so the displayed x-range is increased and therefore the displayed width of the interval changes.

So yes, if the interval would be saved in calibrated coordinates, it would solve the problem.
But what was the confusing thing here is that there was no second line plot anymore in the display item, but it kept changing the displayed x-scale. We only caught this because there was still the "Data 0" label in the top-right corner. Somebody must have dragged a second line plot on top of the live data item some time ago, and the only way to fully revert this seems to delete the live data item, which people usually don't do.

So the bug here is that once you remove all layers but one from a stacked line plot, it does not behave like a line plot that has never been stacked.

AM:
I managed to reproduce it on the simulator. It seems to only appear under very specific conditions, so here is what I did exactly:

1. Click play on the eels camera so that you see a live spectrum
2. Create a interval graphic on the live data item
3. Make a snapshot of the live data item
4. Drag-and-drop the snapshot onto the live data item to create a stacked line plot
5. Change energy offset (with the +eV box at the bottom of the "Simulator Instrument" panel

If you do that, the displayed scale of the data item changes (but not the actual scale that you see in the inspector). But also the displayed width of the interval graphic changes (not the width on the screen but the width it shows when it is selected).

Everything by itself is still correct, but it is just very confusing and will lead to problems if you use that displayed width for example to calibrate the EELS dispersion.

Brow71189, 2020-05-08:
I think this can only be fixed by changing how we save graphics bounds, as it seems you have suggested already in the email conversation above.
In my opinion there is a fundamental decision to make about which object "owns" a graphic. In the current implementation I would argue that they are owned by the display right now but to fix this issue they must be owned by the data (or rather the calibrated coordinate system).

I'm not sure which option is better, maybe we can let the user decide as I think you suggested in your comment in https://github.com/nion-software/nionswift/issues/382#issuecomment-620116872:

> 2+. We should decide whether an interval covers a single data layer or can be useful for multiple layers simultaneously. For instance, the interval could be data layer independent; and applying it to a specific data layer could be an additional operation. This is closest to how it works now.

cmeyer, 2020-05-08:
A couple of quick comments: graphics are definitely staying with displays. We might want to introduce the idea of regions of interest that are associated with either data or reference frames if we go that route and those can be tracked by graphics (this is how early implementations handled it) - but that's a bigger project.

Brow71189, 2020-05-09:
Sure that sounds like a viable option. I mean we will probably need something like this in order to be able to support selecting a specific layer for a graphic in a composite line plot. Or would that work differently?
Either way, I think this issue has to wait until we have some mechanism in place to "link" a graphic to data.