nion-software/nionswift #276: Review cropped_display_xdata scaling

cmeyer, 2019-09-15:
In email from BP/ND on 2019-04-08, they divide a line plot by itself and end up with non 1.0 value.

cmeyer, 2020-04-25:
￼

cmeyer, 2020-04-25:
[image]

ND: Doing division of two four pixel wide line profiles, it gets normalization wrong, note that equation is dividing one by itself.

Brow71189, 2020-04-27:
I can't reproduce this anymore, maybe it is already fixed? Of course I don't have the data that they used, but when I divide some data by itsel I get a straight 1.0:

[image]

cmeyer, 2020-04-27:
Closing: unreproducible.

closed 2020-04-27 by cmeyer