nion-software/nionswift #297: Provide list browser view in data panels

cmeyer, 2019-09-16:
Also include a compact list browser view

cmeyer, 2021-12-13:
Reference [Different Parts of File Explorer Window](https://www.techinfobit.com/learn-about-the-different-parts-of-the-file-explorer-window-in-windows-8/)

cmeyer, 2023-01-17:
Useful additional note from a user (email 2021-11-05):

 3- (!!!!!!!!!...)
the data panel is both a waste of space and resources. It's fine when
you have 3 images. Not when you're dealing with 300 that have very
similar titles and look. This needs to have an option to display data
items in a similar way to Explorer: data titles only, not icons, the
"details" view in Windows speak, with no live display [...]. You
need extra searchable and sortable fields such as: data size, last
modified, data type etc..., that can be toggled on/off. Maybe a 'flag'
field as well, e.g. to highlight whether or not that data item's been
exported and therefore is safely away from Swift. Maybe a colour code or
a colour flag for live/non-live data items. Maybe an option to organise
as trees (or threads), so all processing actions are nested under the
parent data item - saves having to hunt for hours to figure out which
line profile belongs with which image.

cmeyer, 2023-01-17:
Another note from ND same thread:

#3 He's absolutely right that the data window is problematic for finding things, and we have talked about systematic naming et c. Maybe he's right that if one could have a list view with choices about which things to show, then one could see enough items on screen that one does not have to scroll down like crazy even for recent data.

jamesrussell216, 2025-01-22:
I think I may have slightly misinterpreted this to start with, but circling back; the concept is to bring in "View > Details" type option to the Data panel right? From the UI side, how do you see that working with the available real estate (given the aspect ratios of the elements and apologies for the crudeness of the screenshot!!)...

[image]