nion-software/nionswift #306: Binary incompatibility?

ericpre, 2019-10-14:
There is a runtime warning when importing various scientific python libraries, such as scikit-image and scikit-learn. Can it be safely ignored?

### Steps to reproduce:
1. Start Nionswfit
2. Open a python console from the file menu
3. import scikit-image or scikit-learn

### Outcome:
```python
>>> import skimage
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
/opt/miniconda3/envs/nionswift/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192 from C header, got 216 from PyObject
  return f(*args, **kwds)
>>> 
```

cmeyer, 2020-09-25:
I can no longer reproduce this issue. Do you still see it occurring? If not, can we close this?

ericpre, 2020-09-26:
I can't tell because nionswift failt start on this system (linux)
```
Traceback (most recent call last):
  File "/opt/miniconda3/bin/nionswift", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/opt/miniconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3267, in <module>
    @_call_aside
  File "/opt/miniconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3251, in _call_aside
    f(*args, **kwargs)
  File "/opt/miniconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3280, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/opt/miniconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 582, in _build_master
    ws.require(__requires__)
  File "/opt/miniconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 899, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/opt/miniconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 785, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'nionswift' distribution was not found and is required by the application
```

output of conda list:
```
# packages in environment at /opt/miniconda3/envs/nionswift:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
blas                      1.0                         mkl  
ca-certificates           2020.7.22                     0  
certifi                   2020.6.20                py37_0  
freetype                  2.10.2               h5ab3b9f_0  
h5py                      2.10.0           py37hd6299e0_1  
hdf5                      1.10.6               hb1b8bf9_0  
imageio                   2.9.0                      py_0  
intel-openmp              2020.2                      254  
jpeg                      9b                   h024ee3a_2  
lcms2                     2.11                 h396b838_0  
ld_impl_linux-64          2.33.1               h53a641e_7  
libedit                   3.1.20191231         h14c3975_1  
libffi                    3.3                  he6710b0_2  
libgcc-ng                 9.1.0                hdf63c60_0  
libgfortran-ng            7.3.0                hdf63c60_0  
libpng                    1.6.37               hbc83047_0  
libstdcxx-ng              9.1.0                hdf63c60_0  
libtiff                   4.1.0                h2733197_1  
lz4-c                     1.9.2                he6710b0_1  
mkl                       2020.2                      256  
mkl-service               2.3.0            py37he904b0f_0  
mkl_fft                   1.2.0            py37h23d657b_0  
mkl_random                1.1.1            py37h0573a6f_0  
ncurses                   6.2                  he6710b0_1  
niondata                  0.13.11                    py_1    conda-forge
nionswift                 0.15.1                     py_0    conda-forge
nionswift-io              0.14.0                     py_2    conda-forge
nionswift-tool            0.4.5            py37h73816c6_1    conda-forge
nionui                    0.4.1                      py_0    conda-forge
nionutils                 0.3.21                     py_1    conda-forge
numpy                     1.19.1           py37hbc911f0_0  
numpy-base                1.19.1           py37hfa32c7d_0  
olefile                   0.46                     py37_0  
openssl                   1.1.1h               h7b6447c_0  
pillow                    7.2.0            py37hb39fc2d_0  
pip                       20.2.2                   py37_0  
python                    3.7.9                h7579374_0  
python_abi                3.7                     1_cp37m    conda-forge
pytz                      2020.1                     py_0  
readline                  8.0                  h7b6447c_0  
scipy                     1.5.2            py37h0b6359f_0  
setuptools                49.6.0                   py37_0  
six                       1.15.0                     py_0  
sqlite                    3.33.0               h62c20be_0  
tk                        8.6.10               hbc83047_0  
tzlocal                   2.1                      py37_0  
wheel                     0.35.1                     py_0  
xz                        5.2.5                h7b6447c_0  
zlib                      1.2.11               h7b6447c_3  
zstd                      1.4.5                h9ceee32_0 
```

cmeyer, 2020-09-26:
Are you using conda-forge or another version? See #543.

ericpre, 2020-10-02:
As it can be seen from the `conda list` output above, this is using the "defaults" and conda-forge channels - when no channel is specified in the channel column, it means that it comes from the "defaults" channel.

cmeyer, 2021-05-04:
@ericpre, can you revisit this issue again? I've made several changes to how Swift starts in the last several months. Is this still an error?

ericpre, 2021-05-09:
Not occurring anymore.

closed 2021-05-09 by ericpre