nion-software/nionswift #307: Cookiecutter for plugins

ericpre, 2019-10-22:
It would be useful to have a cookiecutter tool to make new plugins. The [cookiecutter for pytest plugin](https://github.com/pytest-dev/cookiecutter-pytest-plugin) is a good example.

cmeyer, 2019-10-22:
Good idea. I looked into this briefly in August but got caught up in other issues.

cmeyer, 2021-11-18:
This is a work in progress, but here is an initial version:

https://github.com/nion-software/cookiecutter-nionswift-plug-in

Once you have installed cookiecutter ([documentation](https://cookiecutter.readthedocs.io/en/stable/README.html)), you can create a new Nion Swift plug-in using the following command:

`cookiecutter gh:nion-software/cookiecutter-nionswift-plug-in`

closed 2022-02-15 by cmeyer