nion-software/nionswift #308: Distribute nionswift and plugin on conda-forge

ericpre, 2019-10-22:
Now that the conda-forge channle is (almost 100%) ABI compatible with the anaconda/defaults channel, it would make sense to distribute nionswift on conda-forge.
Here are a few reasons:
- the release process is fully automated
- no need to use a different channel, which will make things easier to the user when updating packages (nion or dependencies)
- when installing plugin, avoid mixing conda and pip installation: it can sometime break installation if incompatible dependencies are installed and this is not straightforward way to anticipate this, particularly for non expert python user
- it would be easier for plugin developer to release their package on conda-forge by following a recipe example of a nion plugin in conda-forge and most importantly plugin developers do not need to use their own channel.

It would possibly fixed the issue mentioned in #306

cmeyer, 2019-10-22:
This is a good goal - I had made a feedstock recipe for the `nionswift-tool` project, but it's still on the `nion` channel. I will make this [feedstock on conda-forge for all of the sub-projects] a goal for the 0.15 milestone.

ericpre, 2020-01-11:
https://github.com/conda-forge/staged-recipes/pull/10583

cmeyer, 2020-04-10:
Done

closed 2020-04-10 by cmeyer