nion-software/nionswift #312: Nionswift shortcut not working

ericpre, 2019-11-05:
The nionswift shortcut created with `nionswift --alias` fails on some windows system (Win7 or Win10) with the following error message:

[image]

On a different windows system, there is a different error with the linking to the hdf5 library.

In both case, it seems that the shortcut does not activate correctly the conda environment and non compatible libraries are picked up at the startup of nionswift.

In any case, nionswift starts fine from the anaconda prompt, only the shortcut is not working.

cmeyer, 2019-11-06:
Can you attach the output from `conda list` and `conda info` on one of the systems that isn't working? Also, are both systems entirely up-to-date with `conda update --all`?

cmeyer, 2019-11-06:
Also: can you try a minimal installation on one of the problem machines?

```
conda create -n test_swift_launch -c nion nionswift nionswift-tool pywin32
conda activate test_swift_launch
nionswift --alias
```

ericpre, 2019-11-06:
On the windows system with the hdft5 linking error (installer available at https://dev.azure.com/ericpre/hyperspy-bundle/_build/results?buildId=216):

In the base environment:
```

(base) C:\Users\Eric>conda info

     active environment : base
    active env location : C:\Users\Eric\HyperSpy-bundle
            shell level : 1
       user config file : C:\Users\Eric\.condarc
 populated config files : C:\Users\Eric\HyperSpy-bundle\.condarc
                          C:\Users\Eric\.condarc
          conda version : 4.7.12
    conda-build version : not installed
         python version : 3.7.5.final.0
       virtual packages : __cuda=9.0
       base environment : C:\Users\Eric\HyperSpy-bundle  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/nion/win-64
                          https://conda.anaconda.org/nion/noarch
          package cache : C:\Users\Eric\HyperSpy-bundle\pkgs
                          C:\Users\Eric\.conda\pkgs
                          C:\Users\Eric\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\Eric\HyperSpy-bundle\envs
                          C:\Users\Eric\.conda\envs
                          C:\Users\Eric\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.7.12 requests/2.22.0 CPython/3.7.5 Windows/10 Windows/10.0.17134
          administrator : False
             netrc file : None
           offline mode : False

(base) C:\Users\Eric>conda list
# packages in environment at C:\Users\Eric\HyperSpy-bundle:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py37_0
anaconda-client           1.7.2                    py37_0
anaconda-navigator        1.9.7                    py37_0
asn1crypto                1.2.0                    py37_0
astroid                   2.3.2                    py37_0
atomap                    0.2.0                      py_0    conda-forge
attrs                     19.3.0                     py_0
autopep8                  1.4.4                      py_0
babel                     2.7.0                      py_0
backcall                  0.1.0                    py37_0
blas                      1.0                         mkl
bleach                    3.1.0                    py37_0
blosc                     1.16.3               h7bd577a_0
bokeh                     1.4.0                    py37_0
ca-certificates           2019.10.16                    0
certifi                   2019.9.11                py37_0
cffi                      1.13.1           py37h7a1dbc1_0
chardet                   3.0.4                 py37_1003
click                     7.0                      py37_0
cloudpickle               1.2.2                      py_0
clyent                    1.2.2                    py37_1
colorama                  0.4.1                    py37_0
conda                     4.7.12                   py37_0
conda-package-handling    1.6.0            py37h62dcd97_0
console_shortcut          0.1.1                         3
cryptography              2.8              py37h7a1dbc1_0
[... 356 lines cut ...]
numpy-base                1.17.2           py37hc3f5095_0
olefile                   0.46                     py37_0
openssl                   1.1.1d               he774522_3
pillow                    6.2.1            py37hdc69c19_0
pip                       19.3.1                   py37_0
pyreadline                2.1                      py37_1
python                    3.7.5                h8c8aaf0_0
pytz                      2019.3                     py_0
pywin32                   223              py37hfa6e2cd_1
scipy                     1.3.1            py37h29ff71c_0
setuptools                41.6.0                   py37_0
six                       1.12.0                   py37_0
sqlite                    3.30.1               he774522_0
tk                        8.6.8                hfa6e2cd_0
tzlocal                   2.0.0                    py37_0
vc                        14.1                 h0510ff6_4
vs2015_runtime            14.16.27012          hf0eaf9b_0
wheel                     0.33.6                   py37_0
wincertstore              0.2                      py37_0
xz                        5.2.4                h2fa13f4_4
zlib                      1.2.11               h62dcd97_3
zstd                      1.3.7                h508b16e_0

(test_swift_launch) C:\Users\Eric>
```

ericpre, 2019-11-06:
This is the error when calling the shortcut from the command prompt without activating the environment:
<details open><summary>Call shortcut</summary><p>

```
Microsoft Windows [Version 10.0.17134.1069]
(c) 2018 Microsoft Corporation. All rights reserved.

C:\Users\Eric>cd Desktop

C:\Users\Eric\Desktop>"Nion Swift 2019-11-06-191632.lnk"
Using Python from command line: C:\Users\Eric\HyperSpy-bundle
Warning! ***HDF5 library version mismatched error***
The HDF5 header files used to compile this application do not match
the version used by the HDF5 library to which this application is linked.
Data corruption or segmentation faults may occur if the application continues.
This can happen when an application was compiled by one version of HDF5 but
linked with a different version of static or shared HDF5 library.
You should recompile the application or check your shared library related
settings such as 'LD_LIBRARY_PATH'.
You can, at your own risk, disable this warning by setting the environment
variable 'HDF5_DISABLE_VERSION_CHECK' to a value of '1'.
Setting it to 2 or higher will suppress the warning messages totally.
Headers are 1.10.4, library is 1.10.5
        SUMMARY OF THE HDF5 CONFIGURATION
        =================================

General Information:
-------------------
                   HDF5 Version: 1.10.5
                  Configured on: 2019-02-26
                  Configured by: Visual Studio 15 2017 Win64
                    Host system: Windows-10.0.17134
              Uname information: Windows
                       Byte sex: little-endian
             Installation point: C:/Program

Compiling Options:
------------------
                     Build Mode: RelWithDebInfo
              Debugging Symbols:
                        Asserts:
                      Profiling:
             Optimization Level:

Linking Options:
----------------
                      Libraries:
  Statically Linked Executables: OFF
                        LDFLAGS: /machine:x64
                     H5_LDFLAGS:
                     AM_LDFLAGS:
                Extra libraries:
                       Archiver:
                         Ranlib:

Languages:
----------
                              C: yes
                     C Compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Professional/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe 19.16.27027.1
                       CPPFLAGS:
                    H5_CPPFLAGS:
                    AM_CPPFLAGS:
                         CFLAGS:  /DWIN32 /D_WINDOWS  /W3
                      H5_CFLAGS:
                      AM_CFLAGS:
               Shared C Library: YES
               Static C Library: YES

                        Fortran: True
               Fortran Compiler: C:/Program Files (x86)/IntelSWTools/compilers_and_libraries_2018/windows/bin/intel64/ifort.exe 18.0.5.20180823
                  Fortran Flags: /W1 /nologo /fpp /libs:dll /threads
               H5 Fortran Flags:
               AM Fortran Flags:
         Shared Fortran Library: YES
         Static Fortran Library: YES

                            C++: ON
                   C++ Compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Professional/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe 19.16.27027.1
                      C++ Flags: /DWIN32 /D_WINDOWS /W3 /GR /EHsc
                   H5 C++ Flags:
                   AM C++ Flags:
             Shared C++ Library: YES
             Static C++ Library: YES

                            JAVA: True
                   JAVA Compiler: C:/Program Files/Java/jdk-11.0.1/bin/javac.exe

Features:
---------
                   Parallel HDF5: OFF
Parallel Filtered Dataset Writes:
              Large Parallel I/O:
              High-level library: ON
                    Threadsafety: OFF
             Default API mapping: v110
  With deprecated public symbols: ON
          I/O filters (external):  DEFLATE DECODE ENCODE
                             MPE:
                      Direct VFD:
                         dmalloc:
  Packages w/ extra debug output:
                     API Tracing: OFF
            Using memory checker: OFF
 Memory allocation sanity checks: OFF
          Function Stack Tracing: OFF
       Strict File Format Checks: OFF
    Optimization Instrumentation:
Bye...

C:\Users\Eric\Desktop>
```
</p></details>

After activation of the environment, calling the shortcut works fine.

cmeyer, 2019-11-06:
Another item to check: What is your PATH variable set to in Windows? Is there another Python and/or another HDF5 available on PATH?

ericpre, 2019-11-07:
On this system there is no other python distribution but there is indeed the hdf5 library installed and one reason to activate the conda environment is set correctly the PATH among other things.
Is it expected that the shortcut does not activate the environment?

Usually on windows one way to create a shortcut with environment activation is to do something along the line of 
```
%windir%\System32\cmd.exe "/K" PATH_TO_ENVIRONMENT\Scripts\activate & nionswift 
```
I have just tried It on the system with the hdf5 issue and it works fine but I am pretty sure I have tried this on the system with the other issue and it went further but there was another issue... I would need to check when I can get on to this system.

ericpre, 2019-11-07:
> I have just tried It on the system with the hdf5 issue and it works fine but I am pretty sure I have tried this on the system with the other issue and it went further but there was another issue...

Apparently, this is dependent on what is in the _Start in_ entry of the shortcut and changing its default value to another folder seems to make it work.

cmeyer, 2019-11-07:
The developers at conda say that the environment must be activated; and you are indeed seeing a problem that arises when it is not activated. However, my experience shows that as long as PATH is clean, it will most likely work without activating.

Even with a clean path, activating is required for some conda packages, but it's not clear to me which ones (I haven't found one yet). And the `NionSwiftLauncher.exe` does make some attempt to configure the PATH properly.

There is a tool which might help detecting problematic packages. I'm wondering if you can try it on your machine? On my system, it doesn't detect anything.

```
conda install conda-prefix-replacement
cpr detect PATH_TO_CONDA
```

Also, why is HDF5 modifying the PATH? Is there a tool it is providing? For system sanity, it might be better to isolate that PATH modification to its own environment on your machines. Isolating it won't solve the bigger problem, but it might solve your specific issue.

Finally, on my system I needed to use the following in the Shortcut:

```
%windir%\System32\cmd.exe "/K" PATH_TO_BASE_CONDA\Scripts\activate nionswift & PATH_TO_BASE_CONDA\envs\nionswift\Scripts\NionSwiftLauncher\NionSwift.exe PATH_TO_BASE_CONDA\envs\nionswift nionui_app.nionswift
```

Changing this changes the icon and the Console window stays open when Swift exits. There might be solutions for both of these problems, but I haven't explored them yet.

ericpre, 2019-11-08:
> Even with a clean path, activating is required for some conda packages, but it's not clear to me which ones (I haven't found one yet). And the NionSwiftLauncher.exe does make some attempt to configure the PATH properly.

Indeed, and on top of that, this may an issue if some plugins requires activation of environment.

> Also, why is HDF5 modifying the PATH? Is there a tool it is providing? For system sanity, it might be better to isolate that PATH modification to its own environment on your machines. Isolating it won't solve the bigger problem, but it might solve your specific issue.

To make sure that it is clear, the hdf5 library is not installed in a conda environment: it has been installed using the official installer.

cmeyer, 2020-03-28:
Is this still an issue? Can you re-test or close?

ericpre, 2020-03-30:
Which issue are you referring to? It seems that there are several here. Can you please be more precise? What would you like to be tested?

cmeyer, 2020-09-25:
The issue is whether `nionswift --alias` produces an alias that can be used on your system. Can you retest and confirm or close this issue?

ericpre, 2020-09-26:
nionswift installed in a new environment doesn't start on this system:

```
Using Python from command line: C:\ProgramData\Miniconda3\envs\nionswift
Launch time 2020-09-26 15:01:42.393347
Python version 3.7.9 (default, Aug 31 2020, 17:10:11) [MSC v.1916 64 bit (AMD64)]
User interface class QtUserInterface / HostLibProxy
Qt version 5.12.6
Application data: C:\Users\Eric\AppData\Local\Nion\Nion Swift\nionswift_appdata.json
Loading plug-ins from C:\Users\Eric\AppData\Roaming\Nion\Nion Swift\PlugIns
NOT Loading plug-ins from C:\Users\Eric\Documents\Nion\Swift\PlugIns (missing)
Plug-in 'nionswift_plugin.DM_IO' loaded (nionswift_plugin).
Plug-in 'nionswift_plugin.TIFF_IO' loaded (nionswift_plugin).
Plug-in 'nionswift_plugin.none' loaded (nionswift_plugin).
NOT Loading color maps from C:\Users\Eric\AppData\Local\Nion\Nion Swift\Color Maps (missing)
Using existing profile C:\Users\Eric\AppData\Roaming\Nion\Nion Swift\Profile.nsproj
Using cache C:\Users\Eric\AppData\Roaming\Nion\Nion Swift\Profile Cache.nscache
Traceback (most recent call last):
File "C:/ProgramData/Miniconda3/envs/nionswift/Scripts/NionSwiftLauncher\bootstrap.py", line 128, in start
return self.__app.start()
File "C:\ProgramData\Miniconda3\envs\nionswift\lib\site-packages\nion\swift\Application.py", line 173, in start
profile, is_created = self.__establish_profile(profile_path)
File "C:\ProgramData\Miniconda3\envs\nionswift\lib\site-packages\nion\swift\Application.py", line 356, in __establish_profile
profile.read_profile()
File "C:\ProgramData\Miniconda3\envs\nionswift\lib\site-packages\nion\swift\model\Profile.py", line 416, in read_profile
self.storage_system.set_property(self, "version", FileStorageSystem.PROFILE_VERSION)
File "C:\ProgramData\Miniconda3\envs\nionswift\lib\site-packages\nion\swift\model\FileStorageSystem.py", line 318, in set_property
self.__write_properties_if_not_delayed(object)
File "C:\ProgramData\Miniconda3\envs\nionswift\lib\site-packages\nion\swift\model\FileStorageSystem.py", line 247, in __write_properties_if_not_delayed
self._write_item_properties(item)
File "C:\ProgramData\Miniconda3\envs\nionswift\lib\site-packages\nion\swift\model\FileStorageSystem.py", line 253, in _write_item_properties
self._write_properties()
File "C:\ProgramData\Miniconda3\envs\nionswift\lib\site-packages\nion\swift\model\FileStorageSystem.py", line 400, in _write_properties
os.replace(temp_filepath, self.__path)
PermissionError
:
[WinError 5] Access is denied: 'C:\\Users\\Eric\\AppData\\Roaming\\Nion\\Nion Swift\\Profile.temp' -> 'C:\\Users\\Eric\\AppData\\Roaming\\Nion\\Nion Swift\\Profile.nsproj'

```

Output of `conda list`:
```
# packages in environment at C:\ProgramData\Miniconda3\envs\nionswift:
#
# Name                    Version                   Build  Channel
blas                      1.0                         mkl  
ca-certificates           2020.7.22                     0  
certifi                   2020.6.20                py37_0  
freetype                  2.10.2               hd328e21_0  
h5py                      2.10.0           py37h5e291fa_0  
hdf5                      1.10.4               h7ebc959_0  
icc_rt                    2019.0.0             h0cc432a_1  
imageio                   2.9.0                      py_0  
intel-openmp              2020.2                      254  
jpeg                      9b                   hb83a4c4_2  
libpng                    1.6.37               h2a8f88b_0  
libtiff                   4.1.0                h56a325e_1  
lz4-c                     1.9.2                h62dcd97_1  
mkl                       2020.2                      256  
mkl-service               2.3.0            py37hb782905_0  
mkl_fft                   1.2.0            py37h45dec08_0  
mkl_random                1.1.1            py37h47e9c7a_0  
niondata                  0.13.11                    py_1    conda-forge
nionswift                 0.15.1                     py_0    conda-forge
nionswift-io              0.14.0                     py_2    conda-forge
nionswift-tool            0.4.5            py37hdb8e271_1    conda-forge
nionui                    0.4.1                      py_0    conda-forge
nionutils                 0.3.21                     py_1    conda-forge
numpy                     1.19.1           py37h5510c5b_0  
numpy-base                1.19.1           py37ha3acd2a_0  
olefile                   0.46                     py37_0  
openssl                   1.1.1h               he774522_0  
pillow                    7.2.0            py37hcc1f983_0  
pip                       20.2.2                   py37_0  
pyreadline                2.1                      py37_1  
python                    3.7.9                h60c2a47_0  
python_abi                3.7                     1_cp37m    conda-forge
pytz                      2020.1                     py_0  
scipy                     1.5.2            py37h9439919_0  
setuptools                49.6.0                   py37_0  
six                       1.15.0                     py_0  
sqlite                    3.33.0               h2a8f88b_0  
tk                        8.6.10               he774522_0  
tzlocal                   2.1                      py37_0  
vc                        14.1                 h0510ff6_4  
vs2015_runtime            14.16.27012          hf0eaf9b_3  
wheel                     0.35.1                     py_0  
wincertstore              0.2                      py37_0  
xz                        5.2.5                h62dcd97_0  
zlib                      1.2.11               h62dcd97_4  
zstd                      1.4.5                h04227a9_0  

```

cmeyer, 2020-09-26:
That is odd. Are you running Swift as administrator (not recommended) or as regular user (recommended)? I'm on my phone today, and I'll look at the stack trace in detail later, but I don't see any reason for a permission error. Have you restarted your computer lately? I wonder if that file is still open from a previous run of Swift. Or another Python instance or library has it open?

ericpre, 2020-10-02:
This is running from the anaconda command prompt, which is not executed with "run as administrator".
Sill the same after restarting the computer.

cmeyer, 2021-05-04:
@ericpre, can you revisit this issue again too? I've made several changes to how Swift starts in the last several months. Is this still an error?

ericpre, 2021-05-09:
When calling `nionswift --alias`, there is a different error:

```
Traceback (most recent call last):
  File "C:\Users\Eric\HyperSpy-bundle\envs\nionswift\Scripts\nionswift-script.py", line 9, in <module>
    sys.exit(main())
  File "C:\Users\Eric\HyperSpy-bundle\envs\nionswift\lib\site-packages\nion\swift\command.py", line 68, in main
    command.launch(sys.argv)
  File "C:\Users\Eric\HyperSpy-bundle\envs\nionswift\lib\site-packages\nion\nionswift_tool\command.py", line 15, in launch
    from win32com.shell import shell, shellcon
ModuleNotFoundError: No module named 'win32com'
```

cmeyer, 2021-05-09:
Thanks for trying this again.

The `--alias` functionality is primarily intended for use by our service engineers now, with the Windows installer or macOS drag and drop versions as the preferred installation for end users (and, soon, a Linux AppImage).

So to use `--alias`, you need to install an extra library (`pywin32`) that we don't include by default.

Can you `conda install -c conda-forge pywin32` (assuming `conda-forge`) and then try the alias creation again? Thanks.

ericpre, 2021-06-11:
After installing `pywin32`, the shortcut creation is working. It would be good to update the documentation in https://nionswift.readthedocs.io/en/stable/installation.html#running-nion-swift accordingly.
The shortcut itself doesn't work: there are still issue with loading various dll because the conda environment is not activated.

See for example: https://github.com/conda-forge/miniforge_console_shortcut-feedstock/blob/master/recipe/console_shortcut.json, which is similar to what you are using yourself, as you mentioned [above](https://github.com/nion-software/nionswift/issues/312#issuecomment-551198908).

https://github.com/kiwi0fruit/shortcutter, is doing the job very well, maybe it would be worth considering using it or documenting it.

cmeyer, 2021-06-11:
Ok thanks for checking. I updated the documentation slightly and added a link to shortcutter. ded525c09925d50b257ed208ab66a56035013d7a

I'm closing this issue for now.

closed 2021-06-11 by cmeyer