nion-software/nionswift #320: Improve long computations with progress bar and cancel button

cmeyer, 2019-12-18:
Also handle exiting or deleting in middle of computation.

Brow71189, 2021-03-17:
Also handle changing input parameters: When any input parameter of a long-running computation changes the currently running execution should be canceled and re-started with the new input parameter.

kociak, 2021-12-23:
This is a very important point that you both raised indeed. progress bar would be great and cancellation on parameter change. Probably also something simple and clear sign to the user would a blinking element (for example the widget of the source and target, or the focus of them, blinking). This  would make clear that you need to quietly wait for the result and not change the input parameter too quick.

After a chat with colleagues (user and coder) I had also the feeling that being able to have non-automatic, or periodic changes on parameter changes would be nice. Assume you have a 2k*2K image that change on a ms rate, you don't need the e.g FFT to update at ms rate, and some computations will even never complete on this time scale. So, maybe a 10 or 24 computations per second would be largely enough to please the user's eye.

I have also two other points to raise about the computations:
1. In many occasions, you like to try a computation on a subset of the data, and when the parameters seems OK on this subset, launch it on the full data. Typical user case is removing an offset on a spectrum (extracted with a pick operation), and when the offset is well-calculated, launch the subtraction on the whole. More interesting is the fit on typical spectra of a SI, where you can deduce typical fitting parameters guesses, then launch on the full SI. We can do it in two different computations, but maybe this would be cool to be able to change the output of the computation in a deterministic manner
2. One thing I love with computation is the fact that you can keep track of the dependencies. However, for long computations, you may want to not authorize updates, but still keeping explicit the dependencies (easier to navigate in your data). One way is to snapshot the output, but the link is destroyed... Maybe a snapshot with "frozen" dependency would be helpful?