nion-software/nionswift #338: Provide a way for plug-ins to access mouse location and listen to click events

Brow71189, 2020-02-07:
Someting on the lines of `document_controller.courser_position_changed_event`. I can see different ways to implement this, but I guess the most useful and simplest to use way would be to have an event in, say, document controller whose listener function would have the following form:
```
listener_fn(position: Geometry.FloatPoint, display_item: DisplayItem.DisplayItem)
```
For mouse clicks one could add `modifiers` to the arguments to give the plug-in acces to keyboard modifiers, too.

Brow71189, 2020-02-07:
As a sidenote:
Right now you have to do something like this to get the mouse position on a click:
```python
dc = api.application.document_controllers[0]

original_mouse_clicked = dc._document_controller.selected_display_panel.canvas_widget.on_mouse_clicked

def on_mouse_clicked(x, y, modifiers):
    original_mouse_clicked(x, y, modifiers)
    canvas_item = dc._document_controller.selected_display_panel.root_container._RootCanvasItem__mouse_tracking_canvas_item
    if canvas_item:
        canvas_item_point = dc._document_controller.selected_display_panel.root_container.map_to_canvas_item(Geometry.IntPoint(y=y, x=x), canvas_item)
        if hasattr(canvas_item, 'map_widget_to_image'):
            image_point = canvas_item.map_widget_to_image(canvas_item_point)
            # The following line prints the mouse click position in the image in pixels relative to top-left corner
            print(image_point)
   
dc._document_controller.selected_display_panel.canvas_widget.on_mouse_clicked = on_mouse_clicked
```

cmeyer, 2020-02-07:
What is the intended use of this? Mouse cursor handling has an extreme effect on UI responsiveness, so I'm apprehensive to have packages modifying it. Is the intended use something that should go into Swift anyway? Or is there another way to provide a similar functionality?

Brow71189, 2020-02-07:
The use case for this specific example is what the guys in Vienna wanted to do (see the email I forwarded you on Feb 3).
They want to know where in an image the user clicked to select and atom that was found by their algorithm. I guess they are drawing some sort of annotations into the image, since having hundreds of graphics in a data_item is too slow for their application.
I guess an alternative solution would be to speed up handling many graphics in an image. Because then the user would just select one and one could get the desired position from there.

Brow71189, 2020-02-08:
I get that cursor tracking can have a huge impact on UI performance, but how about mouse clicks? Most use cases of this issue would probably be covered by supporting listening to mouse click events.

cmeyer, 2020-06-04:
Partially implemented b2f3adba2dda7cf3340ab385f1d0164e18dcc3e4

cmeyer, 2024-02-25:
See mouse handlers around b66479f66b6fc650be9ebcf9dd36e3ff49919112