nion-software/nionswift #339: Add a "force dock" functionality for panels

Brow71189, 2020-02-07:
Especially on the rather small laptop screens, if you undock a panel in Swift you will have a really hard time docking it again just by hovering over the sides of the window. Maybe we could add a “force dock” button to panels or something similar.

cmeyer, 2020-02-07:
The panel docking functionality is provided by Qt - so this request might need to go upstream.

What platform is this having a problem? Windows and macOS use version 5.14 of Qt - while Linux uses a much older version (5.9 on some versions).

Do you have the same problem with the PyQt or PySide2 backend rather than `nionui-tool` or `nionswift-tool`?

Brow71189, 2020-02-07:
The problem during my HU visit occured on a Mac installation with `nionswift-tool`. But I've also seen it happend on my windows machine.
I understand that this is mostly handled by Qt, but maybe we can still improve the situation.
For example spyder provides a "Lock panes and toolbar" option which does not allow the user to change the arrangement of panels (but you can still resize them). This removes the risk of accidentally undocking a panel. In general, spyder only rearranges panels on drag, but does not undock them. You undock panels via a little "hamburger menu" in the header of each panel (and re-dock it with the same one when it is undocked).
Since they are also using Qt, maybe it is not too hard to use that behavior in Swift.

cmeyer, 2021-03-25:
I've since learned that double-clicking the title bar of a palette will perform "force dock". Closing this issue.

closed 2021-03-25 by cmeyer