nion-software/nionswift #341: `show()` method in built-in python console should return the created data_item

Brow71189, 2020-02-07:


cmeyer, 2021-03-25:
I'm not sure this would be good behavior since `show` may not always create a single data item.

However, perhaps we can just provide a similarly simple method which does return a data item from any type of input data array equivalent.

`show(new_data_item(my_numpy_array))`

Not quite as easy to remember - but I'm not ready to tie the return type of `show` specifically to a data item yet.

Brow71189, 2021-03-25:
Hm ok...but could we just make it return the object(s) it creates? This is anyway a method you want for quick-and-dirty tests on the console so that you can just explore live whatever object it returns. It would just make testing something so much easier if it had that feature. Right now I always end up doing things like:
```
show(something)
# Look for the created object (usually a data item) in the Data panel
# Drag it into a panel
# Press Ctrl + Shift + K
# Get some R-number on the console
do_something(r01.data)
```