nion-software/nionswift #357: Standardized metadata access should also be available on xdata objects

Brow71189, 2020-02-27:
Metadata is stored in the `DataAndMetadata` object. So why do we have to access standardized metadata values via a `DataItem`? Being able to access this on `DataItem`s is good for convenience but it should definitely be availabel for `DataAndMetadata` objects.
For example when you want to use certain acquisition properties like say "binning" in an application that receives `DataAndMetdata` you have to browse through the `metadata` dict manually. It would be nice to be able to use standardized access to such parameters to ensure future compatibility.

https://github.com/nion-software/nionswift/blob/12f619084044082066cd91748bcde361eb884a5d/nion/swift/model/DataItem.py#L1194

cmeyer, 2021-03-25:
See #663 expanded data item

cmeyer, 2025-03-25:
This is already possible. I added docs here 414937c03b8cf0cbf57496877def7c00b27a5a1c.

closed 2025-03-25 by cmeyer