nion-software/nionswift #363: Test fail on Windows

Brow71189, 2020-03-27:
Most of the errors are due to Windows not being able to change a file that is open in a different process, so that is kind of expected. But there are also a bunch of failures, like e.g.:
```
FAIL: test_processing_pick_configures_in_and_out_regions_and_connection (nion.swift.test.DocumentModel_test.TestDocumentModelClass)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\DocumentModel_test.py", line 142, in test_processing_pick_configures_in_and_out_regions_and_connection
    self.assertFalse(numpy.array_equal(pick_data_item.data, d[4, 4, :]))
AssertionError: True is not false
```
Which I don't understand why they fail.

I ran the following commands to run the tests (while being in the `nionswift` repository root directory):
```
conda create -n run_nionswift_tests python=3.7
conda activate run_nionswift_tests
pip install -r test-requirements.txt
python -m unittest discover -p "*_test.py"
```
Doing the exact same thing in "Bash on Ubuntu on Windows" (which is a linux bash you can get from the windows subsystem for linux) with the source code being the exact same causes all tests to pass.

Below is the full output of running the testsuite on Windows:

```
C:\cygwin64\home\Andi\git\nionswift>python -m unittest discover -p "*_test.py"
.......C:\Users\Andi\Miniconda3\envs\run_nionswift_tests\lib\asyncio\base_events.py:626: ResourceWarning: unclosed event loop <_WindowsSelectorEventLoop running=False closed=False debug=False>
  source=self)
ResourceWarning: Enable tracemalloc to get the object allocation traceback
................................................................................F................................................................................................................................................................................................................................................................................................................................................F...............................................F................................................................................................F.......................................................................................................E..............................................................................................................................................xEEEEEE......F...........EE......E..................................EE...EE...E..EE....EE........EExE..E.E..x.............EE..E.E..xE...E.....EE.E.EE..........F.F.........F...F......FF..............................................F...........F..........................................................
======================================================================
ERROR: test_float_to_string_converter_strips_units (nion.swift.test.Inspector_test.TestInspectorClass)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\Inspector_test.py", line 354, in test_float_to_string_converter_strips_units
    self.assertAlmostEqual(converter.convert_back("0,500"), 0.5 / 256)
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\Inspector.py", line 1736, in convert_back
    return calibration.convert_from_calibrated_value(Converter.FloatToStringConverter().convert_back(str)) / data_size
  File "c:\cygwin64\home\andi\git\nionswift\src\nionutils\nion\utils\Converter.py", line 66, in convert_back
    return locale.atof(m.group(0))
  File "C:\Users\Andi\Miniconda3\envs\run_nionswift_tests\lib\locale.py", line 326, in atof
    return func(delocalize(string))
ValueError: could not convert string to float: '0,500'

======================================================================
ERROR: test_auto_migrate_copies_old_data_to_new_library (nion.swift.test.Storage_test.TestStorageClass)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\Storage_test.py", line 3092, in test_auto_migrate_copies_old_data_to_new_library
    with create_temp_profile_context() as profile_context:
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\Storage_test.py", line 119, in create_temp_profile_context
    return TempProfileContext(pathlib.Path.cwd(), no_remove=no_remove)
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\Storage_test.py", line 62, in __init__
    shutil.rmtree(self.workspace_dir)
  File "C:\Users\Andi\Miniconda3\envs\run_nionswift_tests\lib\shutil.py", line 516, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\Andi\Miniconda3\envs\run_nionswift_tests\lib\shutil.py", line 395, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\Andi\Miniconda3\envs\run_nionswift_tests\lib\shutil.py", line 400, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\Andi\Miniconda3\envs\run_nionswift_tests\lib\shutil.py", line 398, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 32] Der Prozess kann nicht auf die Datei zugreifen, da sie von einem anderen Prozess verwendet wird: 'C:\\cygwin64\\home\\Andi\\git\\nionswift\\__Test\\Profiles\\ProfileCache.cache'

======================================================================
ERROR: test_auto_migrate_does_not_overwrite_newer_items (nion.swift.test.Storage_test.TestStorageClass)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\Storage_test.py", line 3238, in test_auto_migrate_does_not_overwrite_newer_items
    with create_temp_profile_context() as profile_context:
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\Storage_test.py", line 119, in create_temp_profile_context
    return TempProfileContext(pathlib.Path.cwd(), no_remove=no_remove)
[... 736 lines cut ...]
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\Symbolic_test.py", line 1260, in test_computation_in_document_recomputes_when_specifier_changes
    self.assertTrue(numpy.array_equal(computed_data_item.data, src_data2 + 1))
AssertionError: False is not true

======================================================================
FAIL: test_resample_produces_correct_data (nion.swift.test.Symbolic_test.TestSymbolicClass)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\Symbolic_test.py", line 833, in test_resample_produces_correct_data
    self.assertTrue(numpy.array_equal(computed_data_item.data, Image.scaled(data_item.data, (5, 4))))
AssertionError: False is not true

======================================================================
FAIL: test_various_expressions_produces_data (nion.swift.test.Symbolic_test.TestSymbolicClass)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\Symbolic_test.py", line 1361, in test_various_expressions_produces_data
    self.assertTrue(numpy.array_equal(computed_data_item.data, data))
AssertionError: False is not true

----------------------------------------------------------------------
Ran 1117 tests in 101.032s

FAILED (failures=13, errors=35, expected failures=4)
```

cmeyer, 2020-03-27:
`test_requirements.txt` needs to match all of the versions in `setup.py`.

I don't like having this stuff in two places... but I didn't see an easy solution at the time I initially set up these tests. Do you want to make a PR on `test_requirements.txt`?

Brow71189, 2020-03-27:
Sure I'll give it a try and make a pull request.

cmeyer, 2020-04-03:
I cleaned up numerous problems revealed but running Windows tests. Can you re-run and see what remains? I think the "0,500" failure in `test_float_to_string_converter_strips_units` may have to do with running on a natively German system.

FYI, all tests pass on my system with commit 590fb648c82210bb3b969b70b713f043f4489702.

Brow71189, 2020-04-06:
I tested again and you're right, the "0.500" failure is specific to my system, because it is set to german but also to use a decimal point instead of the usual comma (this is because AS2 cannot handle having a comma as a decimal separator).

I'm also getting a bunch of new errors like this:
```
======================================================================
ERROR: test_line_profile_with_intervals_does_not_recompute_when_reloaded (nion.swift.test.Storage_test.TestStorageClass)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\Storage_test.py", line 1420, in test_line_profile_with_intervals_does_not_recompute_when_reloaded
    document_model = DocumentModel.DocumentModel(profile=profile_context.create_profile())
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\model\DocumentModel.py", line 594, in __init__
    self.__profile.read_profile()
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\model\Profile.py", line 296, in read_profile
    self.read_project(project_reference)
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\model\Profile.py", line 387, in read_project
    project.prepare_read_project()
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\model\Project.py", line 222, in prepare_read_project
    self._raw_properties = self.__storage_system.read_project_properties()  # combines library and data item properties
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\model\FileStorageSystem.py", line 533, in read_project_properties
    data_items_copy = sorted(properties_copy.get("data_items", list()), key=data_item_created)
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\model\FileStorageSystem.py", line 530, in data_item_created
    earliest_datetime = datetime.datetime.fromtimestamp(time.mktime(time.gmtime(0))).isoformat()
OverflowError: mktime argument out of range
```
This is due to my timezone being UTC+01, which would `time.mktime(time.gmtime(0))` cause to return a negative number and that is apparently not supported by my python implementation (see [https://stackoverflow.com/questions/2518706/python-mktime-overflow-error](https://stackoverflow.com/questions/2518706/python-mktime-overflow-error) for example).
If I set my timezone to UTC (or UTC-08 like yours) these errors disappear.

I'm also still getting the test failures in `Symbolic_test.py`. I'll try to find out what exactly the problem is there next.

Below the full output of my test run (with the decimal point and timezone problem resolved):

```
======================================================================
FAIL: test_recomputing_data_after_cached_data_is_called_gives_correct_result (nion.swift.test.DataItem_test.TestDataItemClass)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\DataItem_test.py", line 1068, in test_recomputing_data_after_cached_data_is_called_gives_correct_result
    self.assertAlmostEqual(inverted_display_item.data_item.data[0, 0], -3.0)
AssertionError: -1.0 != -3.0 within 7 places (2.0 difference)

======================================================================
FAIL: test_recomputing_data_gives_correct_result (nion.swift.test.DataItem_test.TestDataItemClass)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\DataItem_test.py", line 1047, in test_recomputing_data_gives_correct_result
    self.assertAlmostEqual(inverted_display_item.data_item.data[0, 0], -3.0)
AssertionError: -1.0 != -3.0 within 7 places (2.0 difference)

======================================================================
FAIL: test_setting_data_browser_selection_to_multiple_items_via_document_controller_updates_selection_object (nion.swift.test.DataPanel_test.TestDataPanelClass)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\DataPanel_test.py", line 457, in test_setting_data_browser_selection_to_multiple_items_via_document_controller_updates_selection_object
    self.assertEqual({1, 2}, document_controller.selection.indexes)  # items are ordered newest to oldest
AssertionError: Items in the first set but not the second:
1
Items in the second set but not the first:
0

======================================================================
FAIL: test_processing_pick_configures_in_and_out_regions_and_connection (nion.swift.test.DocumentModel_test.TestDocumentModelClass)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\DocumentModel_test.py", line 144, in test_processing_pick_configures_in_and_out_regions_and_connection
    self.assertFalse(numpy.array_equal(pick_data_item.data, d[4, 4, :]))
AssertionError: True is not false

======================================================================
FAIL: test_computation_reconnects_after_reload (nion.swift.test.Storage_test.TestStorageClass)
----------------------------------------------------------------------
[... 65 lines cut ...]
[... cut ...]
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\cygwin64\home\Andi\git\nionswift\nion\swift\test\Symbolic_test.py", line 851, in test_resample_with_data_shape_produces_correct_data
    self.assertTrue(numpy.array_equal(computed_data_item.data, Image.scaled(data_item.data, (5, 4))))
AssertionError: False is not true

----------------------------------------------------------------------
Ran 1129 tests in 96.908s

FAILED (failures=15, expected failures=4)
```

cmeyer, 2020-04-06:
For the date-related tests, can you try changing all uses of `time.gmtime(0)` to `time.mktime(time.localtime(0))` and see if the timing tests pass?

Also, can you run the Python command line and run the commands below and reply to this message with the output?

```
import time
time.gmtime(0)
time.localtime(0)
time.mktime(time.gmtime(0))
time.mktime(time.localtime(0))
```

cmeyer, 2020-04-06:
For the NumPy related failures -- what version of NumPy are you running? Is anything else modified on your system or is everything `master` branch?

If you're going to debug it, you might just focus on `test_computation_using_object_updates_efficiently_when_region_changes` since it uses only integer math and you can just print the two arrays at the end to see what is different. Are they completely different or just one pixel?

cmeyer, 2020-04-06:
For the locale issue, can you run these commands from the Python command line and reply with the output?

```
import locale
locale.localeconv()
locale.atof("0,5")
locale.atof("0.5")
locale.setlocale(locale.LC_ALL, 'de_DE')
locale.localeconv()
locale.atof("0,5")
locale.atof("0.5")
locale.setlocale(locale.LC_ALL, 'fr_FR')
locale.localeconv()
locale.atof("0,5")
locale.atof("0.5")
```

Brow71189, 2020-04-07:
This is with "normal" german settings:
```python
>>> import locale
>>> locale.localeconv()
{'int_curr_symbol': '', 'currency_symbol': '', 'mon_decimal_point': '', 'mon_thousands_sep': '', 'mon_grouping': [], 'positive_sign': '', 'negative_sign': '', 'int_frac_digits': 127, 'frac_digits': 127, 'p_cs_precedes': 127, 'p_sep_by_space': 127, 'n_cs_precedes': 127, 'n_sep_by_space': 127, 'p_sign_posn': 127, 'n_sign_posn': 127, 'decimal_point': '.', 'thousands_sep': '', 'grouping': []}
>>> locale.atof("0,5")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Andi\Miniconda3\lib\locale.py", line 326, in atof
    return func(delocalize(string))
ValueError: could not convert string to float: '0,5'
>>> locale.atof("0.5")
0.5
>>> locale.setlocale(locale.LC_ALL, 'de_DE')
'de_DE'
>>> locale.localeconv()
{'int_curr_symbol': 'EUR', 'currency_symbol': '€', 'mon_decimal_point': ',', 'mon_thousands_sep': '.', 'mon_grouping': [3, 0], 'positive_sign': '', 'negative_sign': '-', 'int_frac_digits': 2, 'frac_digits': 2, 'p_cs_precedes': 0, 'p_sep_by_space': 1, 'n_cs_precedes': 0, 'n_sep_by_space': 1, 'p_sign_posn': 1, 'n_sign_posn': 1, 'decimal_point': ',', 'thousands_sep': "'", 'grouping': [3, 0]}
>>> locale.atof("0,5")
0.5
>>> locale.atof("0.5")
0.5
>>> locale.setlocale(locale.LC_ALL, 'fr_FR')
'fr_FR'
>>> locale.localeconv()
{'int_curr_symbol': 'EUR', 'currency_symbol': '€', 'mon_decimal_point': ',', 'mon_thousands_sep': '\xa0', 'mon_grouping': [3, 0], 'positive_sign': '', 'negative_sign': '-', 'int_frac_digits': 2, 'frac_digits': 2, 'p_cs_precedes': 0, 'p_sep_by_space': 1, 'n_cs_precedes': 0, 'n_sep_by_space': 1, 'p_sign_posn': 1, 'n_sign_posn': 1, 'decimal_point': ',', 'thousands_sep': '\xa0', 'grouping': [3, 0]}
>>> locale.atof("0,5")
0.5
>>> locale.atof("0.5")
0.5
```

And this is with german settings but using "comma" as decimal separator:
```python
>>> import locale
>>> locale.localeconv()
{'int_curr_symbol': '', 'currency_symbol': '', 'mon_decimal_point': '', 'mon_thousands_sep': '', 'mon_grouping': [], 'positive_sign': '', 'negative_sign': '', 'int_frac_digits': 127, 'frac_digits': 127, 'p_cs_precedes': 127, 'p_sep_by_space': 127, 'n_cs_precedes': 127, 'n_sep_by_space': 127, 'p_sign_posn': 127, 'n_sign_posn': 127, 'decimal_point': '.', 'thousands_sep': '', 'grouping': []}
>>> locale.atof("0,5")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Andi\Miniconda3\lib\locale.py", line 326, in atof
    return func(delocalize(string))
ValueError: could not convert string to float: '0,5'
>>> locale.atof("0.5")
0.5
>>> locale.setlocale(locale.LC_ALL, 'de_DE')
'de_DE'
>>> locale.localeconv()
{'int_curr_symbol': 'EUR', 'currency_symbol': '€', 'mon_decimal_point': ',', 'mon_thousands_sep': '.', 'mon_grouping': [3, 0], 'positive_sign': '', 'negative_sign': '-', 'int_frac_digits': 2, 'frac_digits': 2, 'p_cs_precedes': 0, 'p_sep_by_space': 1, 'n_cs_precedes': 0, 'n_sep_by_space': 1, 'p_sign_posn': 1, 'n_sign_posn': 1, 'decimal_point': '.', 'thousands_sep': "'", 'grouping': [3, 0]}
>>> locale.atof("0,5")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Andi\Miniconda3\lib\locale.py", line 326, in atof
    return func(delocalize(string))
ValueError: could not convert string to float: '0,5'
>>> locale.atof("0.5")
0.5
>>> locale.setlocale(locale.LC_ALL, 'fr_FR')
'fr_FR'
>>> locale.localeconv()
{'int_curr_symbol': 'EUR', 'currency_symbol': '€', 'mon_decimal_point': ',', 'mon_thousands_sep': '\xa0', 'mon_grouping': [3, 0], 'positive_sign': '', 'negative_sign': '-', 'int_frac_digits': 2, 'frac_digits': 2, 'p_cs_precedes': 0, 'p_sep_by_space': 1, 'n_cs_precedes': 0, 'n_sep_by_space': 1, 'p_sign_posn': 1, 'n_sign_posn': 1, 'decimal_point': ',', 'thousands_sep': '\xa0', 'grouping': [3, 0]}
>>> locale.atof("0,5")
0.5
>>> locale.atof("0.5")
0.5
```

Brow71189, 2020-04-07:
> For the date-related tests, can you try changing all uses of `time.gmtime(0)` to `time.mktime(time.localtime(0))` and see if the timing tests pass?
>
> Also, can you run the Python command line and run [...]

```python
>>> import time
>>> time.gmtime(0)
time.struct_time(tm_year=1970, tm_mon=1, tm_mday=1, tm_hour=0, tm_min=0, tm_sec=0, tm_wday=3, tm_yday=1, tm_isdst=0)
>>> time.localtime(0)
time.struct_time(tm_year=1970, tm_mon=1, tm_mday=1, tm_hour=1, tm_min=0, tm_sec=0, tm_wday=3, tm_yday=1, tm_isdst=0)
>>> time.mktime(time.gmtime(0))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: mktime argument out of range
>>> time.mktime(time.localtime(0))
0.0
```

Brow71189, 2020-04-07:
> 
> 
> For the NumPy related failures -- what version of NumPy are you running? Is anything else modified on your system or is everything `master` branch?

I'm using numpy 1.18.1. I made sure everything else is on `master` before running the tests, so I don't think anything is modified.
> 
> If you're going to debug it, you might just focus on `test_computation_using_object_updates_efficiently_when_region_changes` since it uses only integer math and you can just print the two arrays at the end to see what is different. Are they completely different or just one pixel?

I had a look at what's going on there and it looks like the result of the computation is not merged back into the target data item. So when you print the results that get compared in the test, the target data item just contains uncropped data.
The reason why the data is not merged back is because the timestamp of the modified data item is not newer than from the original one (in fact they are equal). They are compared here: https://github.com/nion-software/nionswift/blob/master/nion/swift/model/DocumentModel.py#L111
Actually when I run the test multiple times it passes sometimes which is exactly when the two timestamps do not match.
Do you think this could also be because of the limited time resolution of the windows clock?
If I add a sleep of 1 ms before the computation is evaluated in line 101 the tests pass.

closed 2020-05-14 by cmeyer