nion-software/nionswift #370: Crash at start because of 'time.gmtime(0)'  in 'FileStorageSystem.py'

Brow71189, 2020-04-14:
Depending on the timezone that is set and the exact implementation on the platform the error below prevents Swift from starting.

My system: Windows 10 and timezone UTC+01

See also #363, https://docs.python.org/3/library/time.html#time.mktime and https://stackoverflow.com/questions/2518706/python-mktime-overflow-error.

```
Traceback (most recent call last):
File "c:\cygwin64\home\andi\git\nionswift\nion\swift\Application.py", line 166, in start
document_model = DocumentModel.DocumentModel(profile=profile)
File "c:\cygwin64\home\andi\git\nionswift\nion\swift\model\DocumentModel.py", line 591, in __init__
self.__profile.read_profile()
File "c:\cygwin64\home\andi\git\nionswift\nion\swift\model\Profile.py", line 477, in read_profile
project_reference.read_project(self.projects, self.profile_context)
File "c:\cygwin64\home\andi\git\nionswift\nion\swift\model\Profile.py", line 147, in read_project
self.__project.prepare_read_project()  # sets up the uuid, used next.
File "c:\cygwin64\home\andi\git\nionswift\nion\swift\model\Project.py", line 197, in prepare_read_project
self._raw_properties = self.__storage_system.read_project_properties()  # combines library and data item properties
File "c:\cygwin64\home\andi\git\nionswift\nion\swift\model\FileStorageSystem.py", line 533, in read_project_properties
data_items_copy = sorted(properties_copy.get("data_items", list()), key=data_item_created)
File "c:\cygwin64\home\andi\git\nionswift\nion\swift\model\FileStorageSystem.py", line 530, in data_item_created
earliest_datetime = datetime.datetime.fromtimestamp(time.mktime(time.gmtime(0))).isoformat()
OverflowError
:
mktime argument out of range
```

cmeyer, 2020-04-20:
039b6bfcb9d33087ad4e6eaa02c8952ea031a3ca

Can you confirm this is fixed?

Brow71189, 2020-04-20:
I can confirm it is fixed. All tests are passing and Swift starts normally with the latest master branch.

closed 2020-04-20 by cmeyer