nion-software/nionswift #378: Computations that create data with more than 2 dimensions should use "large_format" data_items

Brow71189, 2020-04-23:
https://github.com/nion-software/nionswift/blob/db37dfebc7146497c4b7ad2bb547bf28f0f9cb82/nion/swift/model/DocumentModel.py#L2395

cmeyer, 2020-04-24:
Doesn't `new_data_item` do this already?

Brow71189, 2020-04-25:
`new_data_item` does that only when you pass in data with more than 2 dimensions. Computations create an empty data_item first and populate it later (see the link in my first comment). In this case, the created data_item is NOT of large_format. Since it is impossible to know what kind of data will be put into the data_item, we either have to have the option to specify this explicitly, or make "large_format" the default. Would the latter option have any major disadvantages?

cmeyer, 2025-03-31:
@Brow71189 I think this issue is addressed by this PR 
- https://github.com/nion-software/nionswift/pull/1405

Brow71189, 2025-04-01:
Looks like we can close this then.

We should then probably also update the computations that tried to work around that issue by creating a data item with data of shape (1, 1, 1) intially. This is not urgent, but might be a nice improvement.

closed 2025-04-01 by Brow71189