nion-software/nionswift #381: Line plot channel graphics (cursor) should have more functionality

cmeyer, 2020-04-25:
Requested by OLK 2020-04-21.

cmeyer, 2020-04-25:
What are the use cases? What functionality do we need to add over existing channel graphics? How does this fit within the paradigm of graphics and selected graphics used for targets of actions?

cmeyer, 2020-04-25:
Existing channel graphic:

[image]

Brow71189, 2020-04-27:
Some thoughts on additional functionality:

1. The label should default to the data value at this position. Right now, chances are high that people think the "Add channel graphic" button does not work because the tiny little red line is easy to miss.

2. We need a way for the user to choose to which piece of data the line "belongs" (for multi-line plots). This is also something we have to think through for interval graphics.

3. I think we could use a double-click as a shortcut to add a channel graphic. That's easy to remember and does not happen accidentally too often.

4. We should probably also allow the user to change the appearance of it, i.e. change lenght/width of the line.