nion-software/nionswift #382: Add key to toggle a "cursor" channel graphic on and off.

cmeyer, 2020-04-27:
Also select it and focus so that arrow keys can immediately be used to move it around. Perhaps a key to expand channel to an interval and a key to collapse to a channel.

Need to propose key mappings first.

cmeyer, 2020-04-27:
Cursor should display the value at the cursor position. See #383

Brow71189, 2020-04-27:
What do you think about a double click as a shortcut to add a channel graphic?

See also my comments in #381:

> 
> 
> Some thoughts on additional functionality:
> 
>     1. The label should default to the data value at this position. Right now, chances are high that people think the "Add channel graphic" button does not work because the tiny little red line is easy to miss.
> 
>     2. We need a way for the user to choose to which piece of data the line "belongs" (for multi-line plots). This is also something we have to think through for interval graphics.
> 
>     3. I think we could use a double-click as a shortcut to add a channel graphic. That's easy to remember and does not happen accidentally too often.
> 
>     4. We should probably also allow the user to change the appearance of it, i.e. change lenght/width of the line.

cmeyer, 2020-04-27:
I'm not immediately sold on double-click. Double-clicks are currently used to reset things - like double clicking the axes will reset them; double clicking histogram will reset. Traditionally double-clicks are "select + action", e.g. launch an application. Somehow double clicking to place a cursor doesn't seem consistent with either of these.

Right click: yes. That would be immediately useful.

I'm more inclined to do something like AutoCAD: a keypress (or tool button in the tool palette) to enter 'cursor position mode'; now the channel cursor follows the mouse until the mouse is clicked or the keypress is repeated. If the operation ends while the mouse is outside the line plot, nothing changes (cursor reverts to previous state).

cmeyer, 2020-04-27:
2+. We should decide whether an interval covers a single data layer or can be useful for multiple layers simultaneously. For instance, the interval could be data layer independent; and applying it to a specific data layer could be an additional operation. This is closest to how it works now.