nion-software/nionswift #389: Line plot should provide options for dragging narrow intervals

cmeyer, 2020-04-27:
Perhaps if the cursor in inside the interval it should be a drag?

And if the user holds down ctrl/command key it should always drag? (`modifiers.control`)

Also, we may consider using a different cursor for hand mode on line plots - it's somewhat large and unwieldy. Any thoughts?

cmeyer, 2020-04-27:
Also: 8 pixel hit test for the sides of intervals seems way to large on a Mac. Can we try 6 or even revert to 4? The changing cursor gives plenty of feedback.

cmeyer, 2020-04-27:
Hit testing tolerance distance should maybe be dependent on DPI - however I searched for OS support for this adjustment with no luck.

Brow71189, 2020-04-28:
With the changing cursor we can probably go back to 4 pixels. Or maybe make it asymmetrical, 4 px on the inside of the interval and 6 or 8 on the outside?
Did you see this post on getting screen DPI: https://stackoverflow.com/questions/54271887/calculate-screen-dpi
Since we're using Qt anyway, wouldn't that be a decent solution for us?

Brow71189, 2020-04-28:
> 
> Also, we may consider using a different cursor for hand mode on line plots - it's somewhat large and unwieldy. Any thoughts?
> 

Windows uses this cursor when you hover over something that you can move:

[image]

I guess everyone will understand what it means, but it is not really much smaller than the hand tool.

Inkscape uses this cursor:

[image]

That one I like because it still has a fine pointer that allows you to precisely target a specific area but in incorporates the well-known cross symbol so that it is obvious what it means.

cmeyer, 2020-04-28:
`nionswift-tool` already does all of the DPI translations - and I now realize that it scales mouse clicks too. So just switching to 4 or 5 pixels again would probably be good. I'm still researching if there is a system call for "hit testing tolerance".

cmeyer, 2020-04-28:
Inkscape calls it "cursor tolerance" and [defaults to 8](https://gitlab.com/inkscape/inkscape/-/blob/master/src/ui/dialog/inkscape-preferences.cpp#L1528).

ArcGIS also calls it cursor tolerance and reportedly [defaults to 5](https://gis.stackexchange.com/questions/141806/how-to-adjust-the-tolerance-when-showing-vertices-with-the-v-key-arcgis).

cmeyer, 2020-04-28:
I changed hit testing to use a `ui_settings` for determining cursor tolerance. `ui_settings` gets its value from the hosting application `UserInterface` class.

See 3a539b966f0fc25a09da4250e8b15899f766c910 and https://github.com/nion-software/nionui/commit/a62e617d54c91300b92fd8fbc5f1fc7052f3ae83

The cursor tolerance is now set to 5. However, this bug is still open (force drag and possibly better cursor).

Brow71189, 2020-04-29:
That's a good solution. What do you think about my cursor suggestions?

commit e04bc246 (2020-05-05): Force drag interval graphics when control is held down. Fixes #389.
closed 2020-05-05 by cmeyer