nion-software/nionswift #392: Line plot displays negative data strangely when display on log-y scale

cmeyer, 2020-04-29:
`show(numpy.arange(-2,5,0.01))` then switch to 'Log Y' then drag axis upwards.

[image]

This is going to require a change to our graphing algorithm and we need to be careful not to impact performance. My first thought is that there will need to be some ability to just not plot the negative data - so as we draw from left to right, we will need to track whether we are drawing or not for any given column and generate appropriate draw commands when switch from drawing to not drawing state. This is the part that may affect performance.

cmeyer, 2020-07-30:
@Brow71189  Is this still an open bug?

Brow71189, 2020-08-04:
Just tested it and it is not an issue anymore:

[image]

closed 2020-08-04 by cmeyer