nion-software/nionswift #403: Go to source/dependent in context menu should not show full title

Brow71189, 2020-05-07:
That is one huge context menu:

[image]

cmeyer, 2020-05-09:
This also fits in with making titles more flexible/dynamic. See #32 (dynamic titles).

Brow71189, 2020-05-10:
Yes, but in general the titles in the context menu should not exceed a certain length. I would just could them after a certain number of characters so that the context menu stays under a reasonable width.

cmeyer, 2020-05-12:
I'd like to add support for text truncation (elision), which is truncation based on character width. And it would be nice to truncate from the middle and replace with an ellipsis. There is no perfect solution for truncation (beginning, middle, end), but I tend to think our names have important information at the beginning and end.

I've added support for truncation to `nionui` but this will not be active until the tools update to 0.4.5. It is available in the PySide2/pyQt version immediately.

closed 2020-05-12 by cmeyer

cmeyer, 2020-05-12:
3fc6def83ca880731a5714bc00803772c5fb78f2

commit be367f2e (2020-05-12): Truncate source/dependent menu item titles to 280 pixels. Fixed #403.