nion-software/nionswift #410: Monospace default font too large on Windows.

Brow71189, 2020-05-11:
With the latest changes the panels that use monospace fonts are using the correct font type again, but now the font size is about twice as large as it should be on Windows:
[image]

If I change the default font size from 11pt to 6pt it is fine again. Linux looks fine with 11 pt.

cmeyer, 2020-05-11:
In the Display panel of Settings in Windows 10, there are two scaling values: one under "Make Text Bigger" and one under "Make Everything Bigger". What are those two values set to on your machine?

cmeyer, 2020-05-11:
Also, in `ImageCanvasItem.py` around line 273, there is a line to set the font for the scale marker. Can you experiment with changing that to the same as what is used in the `OutputPanel` and see if it has the same problem?

Brow71189, 2020-05-12:
> 
> 
> I'm not able to reproduce this problem on any of the machines to which I have access.
> 
> In the Display panel of Settings in Windows 10, there are two scaling values: one under "Make Text Bigger" and one under "Make Everything Bigger". What are those two values set to on your machine?
> 
> Can you also click on Additional Display Settings and then Advanced Scaling Settings - do you have "fix scaling for apps" enabled or disabled? Any custom scaling in that same settings panel?
> 
Text scaling is set to 100%, and "all" scaling to 150% on my main monitor.

> Do you have multiple monitors? Is the behavior the same on the 2nd monitor?
> 
Yes I do. The behavior I described is from my main monitor (a 4k screen). My 2nd one is only FullHD, and I just can't use Swift there becuase all scaling is completely off.

> And finally: if you previously had changed "Make Everything Bigger" scaling, have you rebooted since changing it? If not, can you try rebooting?

The overall scaling has always been on 150% on my machine. This is also the default that Windows suggests, because my screen is not that big but has 4k resolution.

Below are some screenshots with different settings:

First my normal setup (Swift on 1st monitor, 150% scaling):
[image]

Then Swift with 100% scaling also on 1st monitor. Here the fonts are abit too small overall (obviously), but I think there are no big differences between the different panels.

[image]

And finally Swift on my secondary monitor (which uses 100% scaling because it is only FullHD). Of course everything is way too big, but there are also no big differences in font size between the different panels:

[image]

That is pretty much unusable, because everything gets scaled for the main monitor (which is a high resolution screen) on start and then stays like this.

Brow71189, 2020-05-12:
> 
> 
> Also, in `ImageCanvasItem.py` around line 273, there is a line to set the font for the scale marker. Can you experiment with changing that to the same as what is used in the `OutputPanel` and see if it has the same problem?

I think I found what is different in the different panels:
The Output  Panel uses "11**pt**" as font size.
All others specify their font size in "**px**", which makes them look normal.

This is actually very strange, I would have expected it to be the other way round, so that you run into problem with different resolutions when you use "px" as a unit.

Brow71189, 2020-05-12:
Another thing I tried: I can make Windows apply the "System High DPI scaling" instead of "Applicatin High DPI scaling". If I use system scaling, font sizes look normal, but Swift looks blurry, so this is also not a good option.

Brow71189, 2020-05-12:
There is a manual page from qt regarding this topic (I'm sure you know that one): https://doc.qt.io/qt-5/highdpi.html
There is a note on the exact thing we are doing in Swift:

> Applications using UI designs with fixed coordinates look small
> The combination of specifying font size in points and other sizes in pixels is particularly problematic because points are independent from the monitor's resolution. For example, suppose we have a frame of 40x20 pixels around the text "hello". If we use a 12pt font, it would look correct on low resolution monitors. But, on high DPI monitors, the frame would be too small, resulting in the text being clipped.

cmeyer, 2020-05-12:
12px monospace / 1.1 spacing looks good on the machines I tried (Windows 150%, Windows 100%, Mac). Can you give it a try on Linux?

When I went to look at it today, my Windows 150% didn't have the latest code and when I got the latest code (12pt monospace) it was obviously a problem.

cmeyer, 2020-05-12:
More specifically: should Linux be 11px or 12px? (Mac looks better with 11px).

Brow71189, 2020-05-12:
Linux should definitely be 12 px. It is still fairly small, but ok. Linux is overall pretty small, but that might be a OS setting thing that I haven't set properly.

cmeyer, 2020-05-12:
Fixed bbf316136fbd0604c59f43c3f1fad82221300691

closed 2020-05-12 by cmeyer