nion-software/nionswift #427: Add ability to define a custom display transfer function

cmeyer, 2020-05-18:
Either packages can supply a predefined transfer function or the user can edit the equation using Python.

ND 2020-05-20:

What I would want is a display calculation, So that I can tweak the formula, that way the data is still there and correct, but something like a log has a tendency to go ugly at the bottom. This would automatically make all possible display transformations available for the advanced user.

Brow71189, 2020-05-18:
Don't we have "Display scripts" available. I think they don't quite work
yet but basic support is there isn't it?

cmeyer, 2020-05-18:
> Don't we have "Display scripts" available. I think they don't quite work yet but basic support is there isn't it? Chris Meyer <notifications@github.com> schrieb am Mo., 18. Mai 2020, 22:52:

That's something different: it's used for a complete replacement of a display (takes data in and produces a drawing context). And I think it's the a wrong path because it's not something we can maintain over the long term. But the idea is still interesting. It just needs to be formalized.