nion-software/nionswift #430: Add ability to map 1d and 2d operations to collections/sequences

cmeyer, 2020-05-22:
e.g. take FFT of every item in collection.

Brow71189, 2020-12-17:
Some thoughts on implementation details for this:

- On the low level you should probably just be able to specify the axes along which to map the 1d/2d operations. Note that even if the "unmapped" operation was intended to be used on the datum dimension(s) this does not neccessarily mean that in a collection/sequence the axis along which you want to map the computation is automatically the collection/sequence axis.
You can imagine a compuation that works on 1D data but you might want to map it to a 1D collection of 2D data by mapping it along the two data axes. An example for a similar case is undistorting a scanned image (for example to negate sample drift effects). You will probably set up the undistort computation on a HAADF image and then apply the same transformation matrix to a 2D collection of either 1D or 2D data. So now you want to map along the data axes and transform the collection axes whereas before you wanted to transform the data axes.

- On the high level or maybe just the UI level you probably don't want to rely on the user to know the differences between sequence/collection/data axes and when they change their "meaning". In the above example of unwarping an image, when setting up the computation the user should be presented with meaningful options for which axes he wants to transform and along which axes to map the computation. A good way for this would be to let the user choose from options like "scan axes" and "camera axes" and maybe "time axis" instead of sequence/collection/data axes. Like this when a user wants to undistort an image to mitigate drift effects, he can just select "Transform Scan coordinates and map along other coordinates" and the computation would do the right thing for both a HAADF image and a Spectrum Image/4D Image.
It might also be that the user wants to get rid of distortions within one camera frame (for example because the projector system instroduces distortions). In this case the user would select "Transform camera coordinates and map along other coordinates" and the computation would do the right things bot for a single camera frame and a 4D Image.

cmeyer, 2021-02-25:
From #158: 

For instance, a collection of 256x256 x 256x256 should have option to calculate FFT based on display (produces a single 256x256 image) or on each datum (producing 256x256 x 256x256 data), etc.

cmeyer, 2021-02-25:
From #173:

Processing requirements should be based on datum rank, not total rank.

Brow71189, 2021-03-17:
We should also consider performance when designing this system. For large datasets ( > tens of GB) it is actually non-trivial to ensure that these mapped operations will finish within a reasonable amount of time. Simply looping over the non-processed dimesnions can be so slow that a relatively simple operation would take hours to complete. As an example take a 4D dataset in our default format, HDF5. This kind of data is saved in chunks of (multiples of) data dimensions which speeds up writing the data when it comes in and also displaying it in our default way. If you want to reduce such a dataset by integrating different dimensions it becomes important how the data is chunked on disk. Integrating the "chunk" dimensions and iterating over the other dimensions will be fast. Doing it the other way though (iterate chunk dimensions, integrate others) will be super slow, because HDF5 will essentially access the whole dataset n times if n is the total number of data points in your iterations dimensions. Execute the following code snipped to see how dramatic this effect is:

```python
import tempfile
import h5py
import numpy
import time

test_data = numpy.ones((64, 64, 64, 64), dtype=numpy.float32)

with tempfile.TemporaryDirectory() as tmpdir:
    f = h5py.File(f'{tmpdir}/myfile.h5', 'a')
    
    dset = f.create_dataset('chunked', shape=test_data.shape, chunks=(1, 1) + test_data.shape[2:], dtype=test_data.dtype)
    
    dset[:] = test_data
    
    # Make sure data is written to disk completely before continuing
    f.flush()
    time.sleep(3)
    
    print('Starting integration along chunk dimensions...')
    t0 = time.perf_counter()
    # Iterate over non-chunks and integrate chunks
    for k in range(test_data.shape[0]):
        for i in range(test_data.shape[1]):
            numpy.sum(dset[k, i])
            
    t1 = time.perf_counter()
    
    print('Starting integration along non-chunk dimensions...')
    # Iterate over chunks and integrate non-chunks
    for k in range(test_data.shape[2]):
        for i in range(test_data.shape[3]):
            numpy.sum(dset[..., k, i])
    t2 = time.perf_counter()
    
    print(f'Timing results (total dataset size: {dset.size*dset.dtype.itemsize/1024**2:.1f} MB):')
    print(f'Integrate chunks:     {t1 - t0:.1f} s\nIntegrate non-chunks: {t2 - t1:.1f} s')        
    
    f.close()
```

On my machine (which uses NVMe SSDs as drives) this prints:
```
Timing results (total dataset size: 64.0 MB):
Integrate chunks:     0.4 s
Integrate non-chunks: 156.7 s
``` 
An alternative is of course to load the whole dataset into memory (e.g. by not iterating in python but simply using `numpy.sum`  to perform the operation in above example. But this has the disadvantage that we are limited to datasets that fit into memory and that there is no way to report progress and to cancel long-running operations.

The conclusion is that for chunked datasets it is extremely important to make sure to access the data correctly. In hpy 3.0 there is a new method `iter_chunks(sel)` which returns an iterator that iterates correctly over the chunks of a dataset given a part of the dataset that you want to iterate over as the argument `sel` (https://docs.h5py.org/en/stable/high/dataset.html#h5py.Dataset.iter_chunks).  I haven't thought this through yet but maybe that would be a way forwards.

cmeyer, 2021-12-23:
How should an operation that takes extra input be handled? For instance, let's say we want to multiply each spectrum in a spectrum image by another 1D image. How would that be expressed in the UI?

One option to consider: user picks or otherwise reduces the data in the way they want; and then they create a template operation on the reduced data; and then they can choose to apply that template to the entire collection.

cmeyer, 2022-05-17:
Additional comment from 2022-04-20 ND:

> Fourier filter to do a 2D Fourier filter of each frame in the video.
> The clean UI for this would simply be to do an FFT (Ctrl+F), and get a
> sequence of FFTs, apply a filter with the usual tool and then have a
> Fourier filtered sequence of the sequence/video. If I have to do some
> Ctrl+Shift+E script edits, it would be less nice but bearable.

cmeyer, 2022-05-21:
Distinguish between a particular piece of data which has a UUID and a stream of data which can also have a UUID.

cmeyer, 2022-12-13:
Another consideration: user may develop a chain of commands that they would like mapped to the original collection. For example, a 4D STEM dataset could have a FFT, window, mask, IFFT, sum applied to each point. How would this be accomplished?

cmeyer, 2023-10-23:
Comment from @tylerharvey: for most operations, if defaulting to the current collection/sequence index, it should do the element processing on the current collection/sequence index rather than defaulting to index 0.

cmeyer, 2025-08-28:
ND 2025-06-13 email

> If I have a sequence and I want to play a Fourier filtered version of it.
> Taking fft followed by Fourier filter and then hitting play on the sequence, gives a static filtered image, independent of slider location.

closed 2026-08-24 by cmeyer