nion-software/nionswift #433: Add table items and functionality and UI

cmeyer, 2020-05-23:
As an example, tuning results could be stored in a table. A sequence of measurements such as line lengths could produce a table. See Pandas for various operations on tables. Tables should be able to be placed in display panels.

Tiomat85, 2026-09-09:
As part of an extra bit of work I have touched on spiking a table to make my life easier and so will move across some of the requirements and design that I put on myself so we can discuss whether it is the right approach for moving across to Swift generally.

Table Component Requirements

Core

Declarative table definition.
Table consumes a collection of strongly-typed row objects.
Lightweight usage API with minimal boilerplate.
Configurable row heights and column widths.
Support for standard and transposed table layouts.

Columns

Each column should support:

Title/header text.
Width configuration.
Value extraction from a row object.
Type-appropriate rendering.
Built-in text and image support.
Optional custom cell rendering.

Future Considerations

Row, column and cell selection.
Filtering.
Sorting.
Column visibility controls.
User-controlled resizing.
Drag-and-drop.
Interactive cell content (for example buttons).
Performance optimisation for large datasets.

cmeyer, 2026-09-09:
I'll expand the body of this issue, but the core of this particular issue is a task to be able able to expand data items to be able to store pandas-like tables in addition to data arrays.

Is the information above about UI or about table functionality once we are able to track it within the project? If the former, let's open an issue in https://github.com/nion-software/nionui/issues for table functionality.