nion-software/nionswift #447: Add metadata editor

cmeyer, 2020-05-24:


Brow71189, 2020-10-12:
see also #551

Brow71189, 2021-01-14:
This would also be very useful for giving the user a way to edit (advanced) configurations for example for cameras. The frequently used parameters are usually present in the UI, but advanced configuration usually involves hunting for some file on the hard disk and editing it. Since configuration usually is in some hierarchical format (like .json or .ini) we could just use the metadata editor UI for this purpose, too.

cmeyer, 2021-01-14:
I like this idea... and to extend it a little... I've been thinking it would be nice for a metadata editor to not only include a raw data editor; but also to allow subsets of the metadata to be controlled by a declarative UI.

cmeyer, 2022-11-04:
- See #551 too