nion-software/nionswift #448: Add surface plot

cmeyer, 2020-05-24:


MattRoyle, 2026-09-02:
Would you be able to add more detail to the ticket if this is a feature still wanted in the software? Was this requested by a user or for a specific use case? For clarity by surface plot do you mean the graph type of a surface plot used to plot 3D data such as the [Matplotlib surface plot](https://matplotlib.org/stable/gallery/mplot3d/surface3d.html)?

cmeyer, 2026-09-02:
The matplotlib link in your comment is the desired feature. This might be a fun project, but my assessment is that it is very low priority. Where it might be useful to generalize the interface to displays; but there is no current request that I am aware of.